carla-simulator/carla #685: Creation of New player vehicle

ptiwari0664, 2018-08-21:
Hello all,

I would like to create a new map in carla along with a new player agent. To start with, I simply duplicate the testmaptruck and tesla model into it. 
[image]

Now I would like to run this tesla vehicle as an player agent (from which i can get position, velocity and collision information). 

Can someone give me any lead on this ???
Thanks in advance.

nsubiron, 2018-08-21:
Hi @ptiwari0664,

Well, that will depend if you want to do it for the new development API (0.9.x) or the stable version (0.8.2).

#### API 0.8.x

Instead of adding a vehicle manually, create _Player Starts_ in the map. When you launch the game, the player vehicle will be spawned in one of these positions. To select an specific vehicle, e.g. Tesla, go to the Game Mode blueprint and change the default pawn from Mustang to Tesla (Alternatively, you can set it in your settings at `Unreal/CarlaUE4/Config/CarlaSettings.ini`).

I would actually recommend using this API for now. For that you should switch to the `stable` branch of our repo, and recompile following the "stable branch" instructions at https://carla.readthedocs.io/en/stable/.

#### API 0.9.x

In this case, in order to be able to control a vehicle it has to be created from the Python API. Take a look at [our blog post](http://carla.org/2018/07/30/release-0.9.0/#python-api) explaining the basics of the new API. If you want to manually control the vehicle take a look at the `manual_control.py` example script. But note that this new API is in a very early stage of development missing a lot of things, like e.g. collision information.

ptiwari0664, 2018-08-22:
Hi Néstor,

Thanks for the help. Yes, I will be shifting to stable version (0.8.x) along with unreal engine 4.19 -> 4.18 as 0.9.x is still in development and it also does not have ROS support.
 
**However I would just like to confirm that whether stable version 0.8.4 will be able to load a newly created map in CarlaEditor or not ?**

nsubiron, 2018-08-22:
If you plan to use the maps with 0.8.4 better create them with that version. Many of our assets depend on the code (e.g. vehicles, traffic lights and signs, weather), if we change some of that code it may break the assets. The way Unreal works you'll be able to open the map but some things may be missing or not working, although it's usually easy to fix but sometimes tedious.

As for the new maps we may add, I don't think we'll have any new map available before the 0.9.x API is stable and supporting ROS.

chengjinjack, 2018-09-19:
hello,@ptiwari0664.
Excuse me, can  you create a new map  and  run it in carla?

ptiwari0664, 2018-09-25:
Yes, using the CarlaEditor you can create a new map however I will advise to use pre-build assets in the new map.

closed 2018-12-01 by stale[bot]