carla-simulator/carla #691: speed / distance unit consistency

Florence-C, 2018-08-22:
Hello, 

I am wondering if there is an issue with the units. 

Indeed I launched the client_example.py, that I slightly modified to have the speed in m/s (just removed the *3.6) , and display the time (fps was 5, so every iteration is 0.2 second).
According to #273, distance is in meters and speed in m/s but in the result below, we can see that between time 7000 and 13000 (ie during 6s), the car travels 29.8 - 16.7 = 13.1 meters, but the speed is 4 m/s. It should have travelled around 24 meters.

Am I doing something wrong ?

[image]

Thanks !

nsubiron, 2018-08-23:
Hi @Florence-C, thanks for the report! we can totally reproduce this and found out that the cause is the current configuration of the physics engine. It is clamped to always keep above 10FPS, thus at 5FPS even if the time-step is reported to be 0.2s, it is actually 0.1s. That's way it looks like reporting twice the speed. If you test with a fixed time-step of 10FPS everything works fine.

I'm opening  a new issue about the physics configuration.

Florence-C, 2018-08-24:
Hello @nsubiron, thanks for the fast reply, and for finding out where the problem comes from !

closed 2018-10-30 by stale[bot]

kraken24, 2019-07-18:
@Florence-C can you tell how did you set constant velocity of 4 m/s??