carla-simulator/carla #697: ERROR: (localhost:2000) failed to read data: timed out when run client_example.py

danxuhk, 2018-08-23:
When run python client_example.py, an issue of failed to read data appears as:
INFO: listening to server localhost:2000
CarlaClient connected
ERROR: (localhost:2000) failed to read data: timed out
CarlaClient connected
ERROR: (localhost:2000) failed to read data: timed out
CarlaClient connected
ERROR: (localhost:2000) failed to read data: timed out
CarlaClient connected
ERROR: (localhost:2000) failed to read data: timed out

What could be the problem?

nsubiron, 2018-08-24:
Try using a different port, it might be that you are connecting to something else.

danxuhk, 2018-08-24:
I am using the port 2000 by default, and I am able to run example.py in PythonAPI. But I couldn't run the sample in PythonClient, showing failed to read data. If example.py could connect to the port 2000, should client_example.py also be able to connect to the same port, right?

nsubiron, 2018-08-24:
Ah ok, I see. _PythonClient_ no longer works with 0.9.0, we are in the middle of a transition to a new API. 

**_PythonClient_ only works with versions prior to 0.9.0, and _PythonAPI_ only works with 0.9.0 and future versions.**

Sorry for the confusion, we are still trying to document all the changes while implementing the new API.

closed 2018-08-24 by danxuhk

jingzhao3200, 2019-01-21:
> Ah ok, I see. _PythonClient_ no longer works with 0.9.0, we are in the middle of a transition to a new API.
>
> **_PythonClient_ only works with versions prior to 0.9.0, and _PythonAPI_ only works with [...]

May I ask what is the difference between _PythonClient_ and _PythonAPI_?
I am trying to implement the elder version python code to the new version 0.9.2, but it cames this error:
ERROR: (localhost:2000) failed to read data: timed out

I have no clues about how to solve this problem.
Thanks!

nsubiron, 2019-01-21:
Prior to 0.9.0 we used three ports, serialized with protobuf except sensor data, the server part of this was in _CarlaServer_ folder. The client-server protocol was very rigid and on any sort of failure it required to reset the current episode.

On the new version we switched to a dual model, rpc + streaming; the control messages are sent via rpc using [rpclib](http://rpclib.net) while the sensor data is streamed with our own streaming library implementation in _LibCarla_. This model allow us to connect several clients simultaneously and be more robust towards connection errors, plus rpc is quite flexible for adding new messages. The client was then moved to _PythonAPI_ which is implemented in C++.

I'm afraid is nearly impossible to make both systems compatible.

jingzhao3200, 2019-01-21:
> Prior to 0.9.0 we used three ports, serialized with protobuf except sensor data, the server part of this was in _CarlaServer_ folder. The client-server protocol was very rigid and on any sort of [...]

Very helpful! Thanks a lot!

hangxu124, 2019-02-07:
Hi, @nsubiron 

when I try to run spawn_npc.py in Carla 0.9.1(and 0.9.3), I got error like this:
`python spawn_npc.py -n 80 -p 3222

destroying 0 actors

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 122, in <module>
    main()
  File "spawn_npc.py", line 68, in main
    world = client.get_world()
RuntimeError: rpc::timeout: Timeout of 2000ms while connecting to 127.0.0.1:3222`

I also try the default 2000 port, some other port which I choose randomly, and got the same error.

I try the command: `sudo ufw status verbose` and then open my port, however, the errors still exist.
`Status: active
Logging: on (low)
Default: deny (incoming), allow (outgoing), disabled (routed)
New profiles: skip

To                         Action      From
--                         ------      ----
2000                       ALLOW IN    Anywhere                  
2001                       ALLOW IN    Anywhere                  
2000/tcp                   ALLOW IN    Anywhere                  
2001/tcp                   ALLOW IN    Anywhere                  
2000                       ALLOW IN    120.0.0.1                 
2001                       ALLOW IN    120.0.0.1                 
2000 (v6)                  ALLOW IN    Anywhere (v6)             
2001 (v6)                  ALLOW IN    Anywhere (v6)             
2000/tcp (v6)              ALLOW IN    Anywhere (v6)             
2001/tcp (v6)              ALLOW IN    Anywhere (v6)  `

Could you please help me about that?

ydjian, 2019-04-05:
I have the same rpc:timeout issue. 

I just follow the instructions to build carla 0.9.5 with Ubuntu 18.04.
On one terminal I ran `make launch`.
On other terminal I ran `python spawn_npc.py` and have the same error messages. 

It seems that the server is not catching the requests. Is there any way to check whether the server is actually listening?

Derekabc, 2019-04-13:
@ydjian have you solved this problem? Thank you

ydjian, 2019-04-22:
Unfortunately nope

frankyyzy, 2019-07-07:
I encountered the similar problem. I am using the mac fork of the carla version and was able to get the simulator working. I am able to scan 2000 and 2001 in my ports but in tutorial.py it just won't connect and timed out. Can someone please help?

analog-cbarber, 2019-07-08:
It's hard to help without more information. Have you looked at the server log?

frankyyzy, 2019-07-08:
I think the server log means that the port 127.0.0.1:2000 is listening and that's what tutorial.py is trying to connect to.
[2019.07.08-18.20.08:391][802]LogNet: GameNetDriver IpNetDriver_6 IpNetDriver listening on port 2000
[2019.07.08-18.20.08:408][802]LogCarla: Error: Can't find player controller!
[2019.07.08-18.20.08:408][802]LogCarlaServer: New episode 'UEDPIE_1_Town03' started
[2019.07.08-18.20.08:408][802]PIE: Play in editor start time for /Game/Carla/Maps/UEDPIE_1_Town03 1.372
[2019.07.08-18.20.08:409][802]LogCarla: == CARLA Settings ==============================================================
[2019.07.08-18.20.08:409][802]LogCarla: Last settings file loaded: 
[2019.07.08-18.20.08:409][802]LogCarla: [CARLA/Server]
[2019.07.08-18.20.08:409][802]LogCarla: Networking = Disabled
[2019.07.08-18.20.08:409][802]LogCarla: World Port = 2000
[2019.07.08-18.20.08:409][802]LogCarla: Server Time-out = 10000 ms
[2019.07.08-18.20.08:409][802]LogCarla: Synchronous Mode = Disabled
[2019.07.08-18.20.08:409][802]LogCarla: Send Non-Player Agents Info = Disabled
[2019.07.08-18.20.08:409][802]LogCarla: Rendering = Enabled
[2019.07.08-18.20.08:409][802]LogCarla: [CARLA/LevelSettings]
[2019.07.08-18.20.08:409][802]LogCarla: Player Vehicle        = Default
[2019.07.08-18.20.08:409][802]LogCarla: Number Of Vehicles    = 5
[2019.07.08-18.20.08:409][802]LogCarla: Number Of Pedestrians = 15
[2019.07.08-18.20.08:409][802]LogCarla: Weather Id = -1
[2019.07.08-18.20.08:409][802]LogCarla: Seed Vehicle Spawner = 123456789
[2019.07.08-18.20.08:409][802]LogCarla: Seed Pedestrian Spawner = 123456789
[2019.07.08-18.20.08:409][802]LogCarla: Two-Wheeled Vehicles = Enabled
[2019.07.08-18.20.08:409][802]LogCarla: Found 0 available weather settings.
[2019.07.08-18.20.08:409][802]LogCarla: [CARLA/QualitySettings]
[2019.07.08-18.20.08:409][802]LogCarla: Quality Level = Epic
[2019.07.08-18.20.08:409][802]LogCarla: [CARLA/Sensor]
[2019.07.08-18.20.08:409][802]LogCarla: Semantic Segmentation = Enabled
[2019.07.08-18.20.08:409][802]LogCarla: ================================================================================
[2019.07.08-18.20.08:512][802]LogInit: FAudioDevice initialized.
[2019.07.08-18.20.08:670][802]LogSlate: Updating window title bar state: overlay mode, drag disabled, window buttons hidden, title bar hidden
[2019.07.08-18.20.08:960][802]LogNet: Browse: 127.0.0.1:2000//Game/Carla/Maps/Town03
[2019.07.08-18.20.08:960][802]LogTemp: Display: ParseSettings for GameNetDriver
[2019.07.08-18.20.08:960][802]LogTemp: Display: ParseSettings for IpNetDriver_7
[2019.07.08-18.20.08:960][802]LogTemp: Display: ParseSettings for PendingNetDriver
[2019.07.08-18.20.08:960][802]LogInit: BSD IPv4/6: Socket queue 32768 / 32768
[2019.07.08-18.20.08:963][802]PacketHandlerLog: Loaded PacketHandler component: Engine.EngineHandlerComponentFactory (StatelessConnectHandlerComponent)
[2019.07.08-18.20.08:964][802]LogNet: Game client on port 2000, rate 10000
[2019.07.08-18.20.09:014][802]PIE: Play in editor start time for /Game/Carla/Maps/Town03 1.978
[2019.07.08-18.20.09:029][802]LogBlueprintUserMessages: Late PlayInEditor Detection: Level '/Game/Carla/Maps/Town03.Town03:PersistentLevel' has LevelScriptBlueprint '/Game/Carla/Maps/Town03.Town03:PersistentLevel.Town03' with GeneratedClass '/Game/Carla/Maps/Town03.Town03_C' with ClassGeneratedBy '/Game/Carla/Maps/Town03.Town03:PersistentLevel.Town03'
[2019.07.08-18.20.09:029][802]LogNet: NotifyAcceptingConnection accepted from: 127.0.0.1:57390
[2019.07.08-18.20.09:029][802]LogHandshake: SendConnectChallenge. Timestamp: 0.090009, Cookie: 244107226208217001080124033131105163199090226016241167159107
[2019.07.08-18.20.09:039][802]LogHandshake: SendChallengeResponse. Timestamp: 0.090009, Cookie: 244107226208217001080124033131105163199090226016241167159107
[2019.07.08-18.20.10:151][803]LogNet: NotifyAcceptingConnection accepted from: 127.0.0.1:57390
[2019.07.08-18.20.10:151][803]LogHandshake: SendChallengeAck. InCookie: 244107226208217001080124033131105163199090226016241167159107
[2019.07.08-18.20.10:151][803]LogNet: Server accepting post-challenge connection from: 127.0.0.1:57390
[2019.07.08-18.20.10:154][803]PacketHandlerLog: Loaded PacketHandler component: Engine.EngineHandlerComponentFactory (StatelessConnectHandlerComponent)
[2019.07.08-18.20.10:154][803]LogNet: NotifyAcceptedConnection: Name: Town03, TimeStamp: 07/08/19 11:20:10, [UNetConnection] RemoteAddr: 127.0.0.1:57390, Name: IpConnection_7, Driver: GameNetDriver IpNetDriver_6, IsServer: YES, PC: NULL, Owner: NULL, UniqueId: INVALID
[2019.07.08-18.20.10:154][803]LogNet: AddClientConnection: Added client connection: [UNetConnection] RemoteAddr: 127.0.0.1:57390, Name: IpConnection_7, Driver: GameNetDriver IpNetDriver_6, IsServer: YES, PC: NULL, Owner: NULL, UniqueId: INVALID
[2019.07.08-18.20.10:156][803]LogNet: UPendingNetGame::SendInitialJoin: Sending hello. [UNetConnection] RemoteAddr: 127.0.0.1:2000, Name: IpConnection_6, Driver: PendingNetDriver IpNetDriver_7, IsServer: NO, PC: NULL, Owner: NULL, UniqueId: INVALID
[2019.07.08-18.20.10:493][804]LogNet: NotifyAcceptingChannel Control 0 server World /Game/Carla/Maps/UEDPIE_1_Town03.Town03: Accepted
[2019.07.08-18.20.10:493][804]LogNet: Remote platform little endian=1
[2019.07.08-18.20.10:493][804]LogNet: This platform little endian=1
[... cut ...]
[2019.07.08-18.20.13:527][806]LogNet: Join request: /Game/Carla/Maps/Town03?Name=Franks-MBP-2-9717B6048F488D85206D82A27F657B84?SplitscreenCount=1
[2019.07.08-18.20.13:528][806]LogOnlineSession: Warning: OSS: No game present to join for session (GameSession)
[2019.07.08-18.20.13:540][806]LogNet: Join succeeded: Franks-MBP-2-9717B60
[2019.07.08-18.20.14:687][808]LogOnlineSession: Warning: OSS: No game present to join for session (GameSession)
[2019.07.08-18.20.28:378][ 31]LogSlate: FSceneViewport::OnFocusLost() reason 0

gengtsh, 2020-08-03:
@ydjian have you solved this problem? Thank you

smmmmi, 2020-09-21:
Still a problem in newer releases 0.9.9

evdoks, 2020-09-21:
> Still a problem in newer releases 0.9.9
@smmmmi Do you experience it on Ubuntu or MacOS? 
I've managed to build Carla and PythonAPI on MacOS, the Carla is running, but Python examples cannot connect to the server and fails with the time-out. 

I can start Unreal Editor with `make launch` (or `make launch-only`), then launch Carla from the editor. I can see, that the UE4Editor process is listening on ports 2000 and 2001. The versions of the server and client can be read and match:

```python
import carla

client = carla.Client(host='127.0.0.1', port=2000)
server_version = client.get_server_version()
client_version = client.get_client_version()
```
However, executing an example (e.g., `spawn_npc.py`) or  trying to load a city with `world = client.load_world('Town03')` results in a time-out:
```
    world = client.load_world('Town03')
RuntimeError: time-out of 60000ms while waiting for the simulator, 
make sure the simulator is ready and connected to 127.0.0.1:2000
```

smmmmi, 2020-09-22:
Thanks for the follow up @evdoks I am however on Ubuntu 18 / Python 3.7, but with the same situation here that Carla is running but the examples time out.
I can however manage to run and spawn blueprints using the apt-get installation but I want it to be built from source to access some files which are not there (carla.client or client_example).

LittlePower97, 2021-05-10:
Hey guys,

I had the same problem, and changing the client.timeout() was fixing it.
`RuntimeError: time-out of 12000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000`
So what I did is that I have a time.sleep(5) at the beginning of my code.

```
import glob
import os
import sys
import random
import math
import time
try:
    sys.path.append(glob.glob('../CARLA_0.9.10/WindowsNoEditor/PythonAPI/carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

try:
    os.startfile(r"C:\Users\slamine\Documents\CARLA_0.9.10\WindowsNoEditor\CarlaUE4.exe")
except IndexError:
    pass

import carla
time.sleep(5)
actor_list = []

try:
    client = carla.Client('localhost', 2000)
    client.set_timeout(10.0)

    world = client.get_world()
    [...]
```

Tell me what you think about it, and if it fixes your Runtime Error or not.
Have a good day !

nimishamanjali, 2021-05-12:
> Hey guys,
>
> I had the same problem, and changing the client.timeout() was fixing it.
> `RuntimeError: time-out of 12000ms while waiting for the simulator, make sure the simulator is ready and connected [...]

Hello could you tell me in which python file you did the modification?

LittlePower97, 2021-05-12:
Hey @nimishamanjali ,

This is a python file I created. But you could add `time.sleep(5)` before `carla.Client('localhost', 2000)` in any code I guess.
On my code, I also open Carla directly from the script, with the following lines :
```
try:
    os.startfile(r"C:\Users\slamine\Documents\CARLA_0.9.10\WindowsNoEditor\CarlaUE4.exe")
except IndexError:
    pass
```
Tell me if you have other questions !

Dilapsky, 2022-02-05:
Carla has an timeout checking for the system. I think if your machine computation power is not strong, your machine may need more time to do computation. Unfortunately, if the computation time exceed the timeout time, this problem happens:
try to use : 

import carla
client = carla.Client('localhost', 2000)
client.set_timeout(100000.0)

https://carla.readthedocs.io/en/latest/core_world/

Dilapsky, 2022-02-05:
> I want to give some advices to guys who can't solve this problem by changing the port. ( Although this problem seems to only appear in the low version, as some teaching platforms are still using the low version of Carla as a teaching tool, I think the solution to this problem is still meaningful. )
> 
> I am using the version 0.8.4.
> 
> If you meet the 'ERROR : (localhost:2000) failed to read data: timed out', you could try to run the 'manual_control.py' in folder '/PythonClient'. Then, if everything goes well, this means there is no problem with the TCP connection, so the problem may be caused by the computer ​which running slowly. Carla is a big project, it reads huge amounts of data from the simulator, so the prog may briefly be unresponsive while reading data or preparing for an episode (especially when you have a mechanical hard disk). To solve this problem, you just need to open the 'client.py' in folder '/PythonClient/Carla', and change 'timeout' to an appropriate value (I set timeout=150), then your computer has plenty of time to read data and run programs​. Hopefully this will be useful :)

Right way to solve my problem.

amirBgazand, 2024-08-29:
I had the same issue, and every time I get to this problem I just change the port to 3000, and amazingly, it works well.
just run the carlaUE4 by
CarlaUE4.exe -carla-rpc-port=3000
and in your code change the port to 3000 
client = carla.Client('127.0.0.1',3000)