carla-simulator/carla #698: Compiling carla for python 3

lburzawa, 2018-08-24:
Is it feasible to compile carla for python 3? I tried to run "make package" changing "--py2" to "--py3" in Linux.mk but kept getting errors. It looks like "PyBuffer_FromMemory" in PythonAPI/source/libcarla/Actor.cpp is specific to python 2, correct me if I'm wrong.

I was using 0.9.0 branch with Ubuntu 18.04.

iFuSiiOnzZ, 2018-08-24:
Hi

In Python 3 the function `PyBuffer_FromMemory` is no more available, it has to be replaced with `PyMemoryView_FromMemory`. So to make it work with Python 3 replace the line with `auto *ptr = PyMemoryView_FromMemory(reinterpret_cast<char *>(self.GetData()), self.GetSize(), PyBUF_READ);`. In future we will make it work for both, Python 2 and 3.

nsubiron, 2018-08-24:
Also we need to compile Boost.Python for Python 3 and make sure to link against the version we compiled.

analog-cbarber, 2018-08-24:
I think you may need to build separately for each Python 3.x version (i.e. 3.5, 3.6, 3.7).

Supposedly there is some way to build boost-python for multiple versions at a time but
I have been unable to figure out how to do it.

hdy117, 2018-08-27:
In folder Util/BuiildTools/BuildPythonAPI.sh, at line77, I just comment out python2 or python3 according to my need. this could fix the problem temperory.

if ${BUILD_FOR_PYTHON2} ; then

  log "Building Python API for Python 2."

  /usr/bin/env python2 setup.py bdist_egg

fi

#if ${BUILD_FOR_PYTHON3} ; then

#  log "Building Python API for Python 3."

#  /usr/bin/env python3 setup.py bdist_egg

#fi