carla-simulator/carla #699: Question about extract steering angle from manul control

NYJiaxing, 2018-08-24:
Hello All,

I have a question about extract the steering angle when I manual control the vehicle. From other contributer, I know that the range of the manual control by the keyboard is **[-1, +1]** but this is not actual angle value that I can save for further learning and output. Secondly, I just fully read the paper: CARLA: An Open Urban Driving Simulator, in the article, the MP method mentioned that the steering angle controled via **PID algorithm**, so Ibelieve there must be an two digital number that I can expect lol. 

Could you please tell me where to get this angle? Like the 

measurements, sensor_data = self.client.read_data() 

function in the client class.

Thank you so much for your reply.

p-schulz, 2018-08-31:
@NYJiaxing 
you can easily create a control message, where you can **set** a steering value [-1, +1]. The actual steering angle depends on the selected vehicle, you can get the maximum angles in the front wheel blueprint of a vehicle (default Mustang is 70°). 
But I don't know if there's a field for retrieving the steering angle while manually controlling a player.

link-er, 2018-10-01:
@felipecode  @marcgpuig  @p-schulz  my question is related to the current issue - do I understand correctly that if I want to transform steering angle to degrees value I should simply multiply it by 70 (at least for default Mustang)? And how can I access the values for other cars (where is the Blueprint information)? Thanks!

chengjinjack, 2018-10-17:
Hello, everyone!
How can I get the steering angle?

closed 2018-12-23 by stale[bot]

sheelabhadra, 2019-09-18:
Hi @JackRen88, were you able to figure out a way to obtain the current wheel angle?