carla-simulator/carla #702: [0.9.0] avaliable attribute list

hdy117, 2018-08-27:
Thank you for your brilliant work.
Now I am using 0.9.0, I am trying to set attribute of camera sensor. But I don't know what key word to use.
I think it would be great if you could make a list of attribute of all actors, blueprints.
Or, you can just tell me how to check, I can check it by myself.

hdy117, 2018-08-27:
Vehicle List, If you are interested.
INFO: [a2, audi, vehicle]
INFO: [chevrolet, impala, vehicle]
INFO: [police, dodgecharge, vehicle]
INFO: [wranglerrubicon, jeep, vehicle]
INFO: [isetta, bmw, vehicle]
INFO: [model3, tesla, vehicle]
INFO: [prius, toyota, vehicle]
INFO: [ford, mustang, vehicle]
INFO: [micra, nissan, vehicle]
INFO: [grandtourer, bmw, vehicle]
INFO: [tt, audi, vehicle]
INFO: [t2, volkswagen, vehicle]
INFO: [mercedesbenz, coupe, vehicle]
INFO: [nissan, patrol, vehicle]
INFO: [c3, citroen, vehicle]
INFO: [leon, seat, vehicle]
INFO: [carlacola, carlamotors, vehicle]
INFO: [mini, cooperst, vehicle]

hdy117, 2018-08-27:
Carmera Sensor attribute list, if you are interested:

     image_size_x
     image_size_y
     fov
     post_processing

nsubiron, 2018-08-27:
Hi @hdy117,

Thanks for sharing the info! and sorry for the problem of getting your issue duplicated a million times :sweat_smile:

Yes indeed, we didn't expose any method to retrieve the available attributes in the Python API. I will add a method to retrieve them.

PR #736 (2018-09-06): Allow iterating attributes of an Actor's Blueprint
#### Description

Added a way of iterating all the attributes of an actor blueprint, the following code now works

```py
for blueprint in blueprint_library:
    print(blueprint)
    for attribute in blueprint:
        print('  - %s' % attribute)
```

Also does minor improvements to the Python prints of blueprint related objects.

Fixes #702.

#### Where has this been tested?

  * **Platform(s):** Ubuntu 16.04
  * **Python version(s):** Python 2.7
  * **Unreal Engine version(s):** 4.19

---
This change is [[image]](https://reviewable.io/reviews [... cut]
closed 2018-09-06 by nsubiron