carla-simulator/carla #733: Error while running bash CarlaUE4.sh after installing NVIDIA Driver

kk2491, 2018-08-28:
Hi Guys,

I recently started working on Carla and quite new to this project. Please help me to get started.

OS : Ubuntu 16.04
GPU : NVIDIA GEFORCE 920 MX

Background : 
1. Downloaded the compiled version from [Github link](https://github.com/carla-simulator/carla/blob/master/Docs/download.md)
2. Was able to successfully run Carla using **bash CarlaUE4.sh**
Please note that below 2 steps were done without NVIDIA drivers installed in the machine.

3. There was lag in the car movement, hence I decided to install the NVIDIA drivers on Ubuntu. Followed the following procedure.

```
     sudo add-apt-repository ppa:graphics-drivers/ppa
     sudo apt update
     sudo apt install nvidia-387
```
4. After installing the drivers I am not able to run Carla simulator as before. Getting the below error while running **bash CarlaUE4.sh**.

```
[2018.08.28-19.40.07:682][  0]LogInit: Error: Linux_PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'
[2018.08.28-19.40.07:682][  0]LogRHI: Error: OpenGL 4.3 not supported by driver
[2018.08.28-19.40.07:732][  0]LogInit: Error: Linux_PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'
```
Error screenshot

[image]

Kindly help me to fix this issue also let me know if you need more details from my end to get the right solution.

Thank you,
KK

marcgpuig, 2018-08-29:
Hi @kk2491.
I don't really know if your installation was ok. I used the Ubuntu's graphical interface to do it, and it worked fine for me. Also, did you tried to use the mentioned `-opengl3` to run it?

We've seen this problem on some more users and we really don't know so much about it. It's something with your drivers and Unreal, not really about CARLA. Sorry about that :(

Cheers.

kk2491, 2018-08-30:
@marcgpuig Thanks for the reply.

I re-installed the NVIDIA driver, now it is working fine.

However still the simulation is bit slow and lagging.

Thank you,
KK

nsubiron, 2018-09-05:
Linking #757.

kk2491, 2019-01-23:
@geekhut I had to reinstall the NVIDIA driver with the latest version.

geekhut, 2019-01-23:
@kk2491 thanks.

pleasenott, 2024-06-06:
Hello, I run ./CarlaUE4.sh but just get this.
```
GameThread timed out waiting for RenderThread after 60.00 secs
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=139728 LargeMemoryPoolOffset=270864 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```
and I find that your UE4 report so much?
which parameters can help me get more message .