carla-simulator/carla #734: Problem while loading CARLA server

eds89, 2018-08-29:
Hello,

I'm not being able to run CARLA server. The server connection closes 1-2 seconds after being started.  Why does that happen? The ini settings file seems to be loaded correctly as the server correctly binds to the port number specified in it, which in my case is 3997. I'm using the following command line:

``
CarlaUE4.exe -carla-server /Game/Maps/Town01  -carla-settings=settings.ini -benchmark -windowed -resX=480 -resY=240
``

Please see attached pictures.

[image]

[image]

Regards,
Eduardo

nsubiron, 2018-09-03:
Hi @eds89,

Make sure to use the same version of the client and the simulator and only one simulator running in the same port. Check also that no other application is using that port, or the firewall is not blocking it.

eds89, 2018-09-03:
Hi. How would we go about checking the firewall is not blocking it?

eds89, 2018-09-04:
Where is CARLA's event log?

nsubiron, 2018-09-05:
The logs are saved under the same folder as the exe at `CarlaUE4/Saved/` (not 100% sure in Windows is the same path, but otherwise something similar).

closed 2018-11-11 by stale[bot]