carla-simulator/carla #735: Error when "make launch"

jianghe01, 2018-08-29:
i try to build carla 0.9.0 on linux. when i run "make launch", it failed with error : 

Error: UnrealHeaderTool failed for target 'CarlaUE4Editor' (platform: Linux, module info: /home/jianghe01/carla/Unreal/CarlaUE4/Intermediate/Build/Linux/B4D820EA/CarlaUE4Editor/Development/CarlaUE4Editor.uhtmanifest, exit code: OtherCompilationError (5)).
Makefile:620: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/jianghe01/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:9: recipe for target 'launch' failed
make: *** [launch] Error 2

nsubiron, 2018-08-30:
Hi @jianghe01,

Make sure you are compiling with UE4.19 and you follow all the steps in [How to build on Linux](https://carla.readthedocs.io/en/latest/how_to_build_on_linux/).

If the error persist, maybe you can share your whole log, because the actual error occurred in the lines before what you shared.

jianghe01, 2018-08-30:
i try to compile with UE4.19 but when i run "make launch", it stop at
Error: CDO Constructor (CarlaMapGenerator): Failed to find StaticMesh'/Game/Carla/Static/RoadLines/St_Road_XCross_LaneMarking.St_Road_XCross_LaneMarking'

jianghe01, 2018-08-30:
@nsubiron

jianghe01, 2018-08-30:
carlaUE4 EDITOR stop at initializing... 73%

nsubiron, 2018-08-30:
Have you downloaded the assets?

jianghe01, 2018-08-30:
@nsubiron u mean the Content? yes. i download it manually and extract it to carla/Unreal/CarlaUE4/Content/. Include "Blueprints", "LICENSE", "Maps/" and "Static/"

jianghe01, 2018-08-31:
@marcgpuig Oh i see, thank you.

closed 2018-08-31 by marcgpuig

chankim, 2018-09-06:
Hi, all, I have the same issue. I've followed https://carla.readthedocs.io/en/latest/how_to_build_on_linux/ . I have installed llvm-5.0 and my clang is clang5.0.0 and alternatives is set to use clang-5.0. But the log says I'm runnnig clang 3.5.1. What's wrong? (I first followed the linx build instructino which was set to use Unreal 4.18 but changed to new instruction above and doing it.) 

ckim@kimchan:~/CARLA/carla$ make launch
Setup.sh: llvm-5.0 already installed.
Setup.sh: boost-1.67.0 already installed.
Setup.sh: Retrieving rpclib.
Cloning into 'rpclib-2.2.1-source'...
remote: Counting objects: 1447, done.
remote: Compressing objects: 100% (1137/1137), done.
remote: Total 1447 (delta 682), reused 587 (delta 277), pack-reused 0
Receiving objects: 100% (1447/1447), 1.81 MiB | 106.00 KiB/s, done.
Resolving deltas: 100% (682/682), done.
Checking connectivity... done.
Note: checking out '3b00c4ccf480b9f9569b1d064e7a3b43585b8dfd'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Setup.sh: Building rpclib with libc++.
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.9.0 or higher is required.  You are running version 3.5.1

-- Configuring incomplete, errors occurred!
Util/BuildTools/Linux.mk:54: recipe for target 'setup' failed
make: *** [setup] Error 1

darld, 2021-04-23:
run make launch

error:
#include "PhysicsEngine/BodySetupEnums.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
5 warnings and 3 errors generated.
Makefile:315: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/xxxx/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:83: recipe for target 'CarlaUE4Editor' failed
make: *** [CarlaUE4Editor] Error 2

Anyone have same experience?

DokurOmkar, 2021-06-25:
@darld  I am having the same error too. Were you able to fix it?

Liuzy0908, 2021-10-01:
> run make launch
>
> error: #include "PhysicsEngine/BodySetupEnums.h" ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 5 warnings and 3 errors generated. Makefile:315: recipe for target 'CarlaUE4Editor' failed make[1]: [...]

I fixed the problem. You should make sure the update.bat file runs completely! Right?

Liuzy0908, 2021-10-01:
> @darld I am having the same error too. Were you able to fix it?

Hello, how did you solve this problem? Sincerely thank you.

Liuzy0908, 2021-10-01:
> run make launch
>
> error: #include "PhysicsEngine/BodySetupEnums.h" ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 5 warnings and 3 errors generated. Makefile:315: recipe for target 'CarlaUE4Editor' failed make[1]: [...]

> @darld I am having the same error too. Were you able to fix it?

Hello, how did you solve this problem? Sincerely thank you.

ZJH-Scholar, 2022-02-06:
I have the same error when I 'make launch', and here is the message:

Makefile:315: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/jiahao/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

noobC7, 2022-03-26:
the same error here, anyone can help? thanks a lot

...error: misleading indentation; statement is not part of the previous 'if' [-Werror,-Wmisleading-indentation]...
...error: no member named 'GameDir' in 'FPaths'...
...fatal error: 'PhysicsEngine/BodySetupEnums.h' file not found...
[19/24] Link (lld) libUE4Editor-OnlineSubsystemUtils.so
[20/24] Link (lld) libUE4Editor-OnlineBlueprintSupport.so
[21/24] Link (lld) libUE4Editor-OnlineSubsystemNull.so
make[1]: *** [CarlaUE4Editor] Error 6
make[1]: Leaving directory `/home/cplus7/carla/Unreal/CarlaUE4'
make: *** [launch] Error 2

darld, 2022-09-20:
@Liuzy0908 @DokurOmkar 
https://stackoverflow.com/posts/57642085/revisions
install python3.7 and update-alternative

imsrinin, 2022-11-05:
I am installing Carla in Ubuntu 20.4 using UE 4.26 (python 3.9)

when I'm trying to compile the server using the command 'make launch' I get the following error:

make[1]: *** [Makefile:315: CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/snaray23/carla/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:7: launch] Error 2

can anyone help me out?

imsrinin, 2022-11-05:
@darld is there no way to build Carla using the latest python version?

ShunlaiC, 2024-03-25:
make[1]: *** [Makefile:315: CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/snaray23/carla/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:7: launch] Error 2

@imsrinin same problem, have you solved it？