carla-simulator/carla #737: ImportError: No module named carla_ros_bridge.bridge

chengjinjack, 2018-08-30:
Hi. I am ready to use carla_ros_bridge on carla 0.8.4.When I test the installation,running `rostest carla_ros_bridge ros_bridge_client.test` ,appear as follows:

`
[ROSUNIT] Outputting test results to /home/rcj_robot/.ros/test_results/carla_ros_bridge/rostest-test_ros_bridge_client.xml
Traceback (most recent call last):
  File "/home/rcj_robot/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/client.py", line 9, in <module>
    from carla_ros_bridge.bridge import CarlaRosBridge
ImportError: No module named carla_ros_bridge.bridge
[Testcase: testtestTopics] ... ERROR!
ERROR: max time [20.0s] allotted for test [testTopics] of type [rostest/publishtest]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0
`
System configuration:
ubuntu16.04
python2.7

How can I solve this problem?

danil-tolkachev, 2018-08-30:
Hello @chengjinjack again!
Did you solve this issue? In #582 you talked about similar (but not the same) error.

chengjinjack, 2018-09-01:
@danil-tolkachev 
Yes. I have solved this issue.Thanks!

closed 2018-09-04 by marcgpuig

ChiZhang95, 2018-09-13:
Hi @chengjinjack 
Could you explain how did you solve this problem? I met the same error. Thanks in advance.

chengjinjack, 2018-09-13:
@ChiZhang95 
Hi. 
When I run intot the issue,I uninstalled  protobuf  ,then  re-install it .As follows:
`1.sudo apt-get remove python-protobuf
2.sudo pip2 uninstall protobuf .
3.sudo pip2 install  protobuf
`
Hope it helps you！

ChiZhang95, 2018-09-13:
Hi @chengjinjack , Thanks very much. But it seems the error is still there. 
My system configuration is Ubuntu 16.04 and python 2.7.12.