carla-simulator/carla #738: weird 2D bounding boxes

danxuhk, 2018-08-30:


mhusseinsh, 2018-08-31:
@danxuhk hello Dan .. can you share how did you get the vertices of the projection and bounding boxes ?

marcgpuig, 2018-08-31:
Hi @danxuhk.
This is weird. Can I see your transformation code? Also, the second car (from the right) does not match with the bounding box.

danxuhk, 2018-08-31:
@marcgpuig Do you have a sample code for calculating the 3D world coordinates for the 8 vertex points of the 3D bounding boxes? I can compare with mine to see if that is correct...

marcgpuig, 2018-08-31:
@danxuhk Sure!
Here https://github.com/carla-simulator/carla/issues/314#issuecomment-379328792 I've provided such a code, it works with 0.8.1 but you can get the point!
The code you provided to me is for the vehicles, can you provide to me the pedestrian code?

danxuhk, 2018-08-31:
@marcgpuig The code at #314 is for the transformation from the 3d world coordinates to 2d the image plane, and I think I have done it correctly. The problem is how we obtain the 3d coordinates for the 8 vertex points of the 3D bounding box. How I could use the agent.vehicle.bounding_box.transform, agent.vehicle.bounding_box.extent, and the agent.vehicle.translation.location to calculate the 3d coordinates...

danxuhk, 2018-09-03:
@marcgpuig Hi Marc, any idea about my issue? I have been entangled in this for a long time. Your help would be appreciated...

danxuhk, 2018-09-04:
@marcgpuig I tried this, and it does not solve my issue. Actually the function of transform_points just performs the operation of vehicle_transform.matrix * points. I checked inside the function. It seems that when the pedestrian changes the orientation the bounding box starts to be wrong, which probably means that the transformation applied does not do the work, really weird...

marcgpuig, 2018-09-04:
@danxuhk How are you testing that these transformations are wrong?

danxuhk, 2018-09-04:
@marcgpuig I run a video, and I plot all the bounding boxes on each frame. Then I can observe each bounding box with the corresponding object frame by frame. I see that when the vehicle turn, the bounding box starts to be wrong, for example, only wrap a small part of the vehicle, meaning that very probably the bounding box is not transformed correctly.

marcgpuig, 2018-09-04:
@danxuhk 
Weird 😕
The position (`agent.vehicle.bounding_box.transform.location`) stills correct right?
I'll try to reproduce it as soon as I can, thanks for the information.

danxuhk, 2018-09-04:
@marcgpuig Yes Marc. The position is correct. It moves with the object moving...

mhusseinsh, 2018-09-06:
@marcgpuig 
Hello Marc,
I tried to adapt your client to show bounding boxes in my manualcontrol also, with 0.8.4 release, and this is what I get

[image]

mhusseinsh, 2018-09-08:
sorry @marcgpuig
do you have any idea why is this happening ?

marcgpuig, 2018-09-10:
@mhusseinsh what is wrong with your output?

mhusseinsh, 2018-09-10:
@marcgpuig so firstly it doesn't visualize the bounding boxes, only the points
but the most important thing, as you can see, that the points are visualized for the pedestrian only, and the car for example does not have this.

mhusseinsh, 2018-09-10:
@marcgpuig ok i managed to solve the problem

marcgpuig, 2018-09-10:
@mhusseinsh 
> so firstly it doesn't visualize the bounding boxes, only the points

Yes, it is exacly how I made the script as you can see in the comments I've made.

mhusseinsh, 2018-09-10:
@marcgpuig yes I saw it now, thanks (Y)
but here the points will be available for all objects which are in front of the vehicle, even which are hidden behind buildings or so .. I mean like not only the objects which are available in the scene

marcgpuig, 2018-09-10:
@mhusseinsh this issue is not about the occlusion topic and it's not implemented in the code I've provided.

mhusseinsh, 2018-09-12:
I did it like that, and it is showing 8 points around the pedestrian, but as I understand that @danxuhk 's problem was with 2D, right ?
```
            for agent in self._measurements.non_player_agents:
                if agent.HasField('pedestrian'):
                    # get the needed transformations
                    # remember to explicitly make it Transform() so you can use transform_points()
                    pedestrian_transform = Transform(agent.pedestrian.transform)
                    bbox_transform = Transform(agent.pedestrian.bounding_box.transform)

                    # get the box extent
                    ext = agent.pedestrian.bounding_box.extent
                    # 8 bounding box vertices relative to (0,0,0)
                    bbox = np.array([
                        [  ext.x,   ext.y,   ext.z],
                        [- ext.x,   ext.y,   ext.z],
                        [  ext.x, - ext.y,   ext.z],
                        [- ext.x, - ext.y,   ext.z],
                        [  ext.x,   ext.y, - ext.z],
                        [- ext.x,   ext.y, - ext.z],
                        [  ext.x, - ext.y, - ext.z],
                        [- ext.x, - ext.y, - ext.z]
                    ])

                    # transform the vertices respect to the bounding box transform
                    bbox = bbox_transform.transform_points(bbox)

                    # the bounding box transform is respect to the pedestrian transform
                    # so let's transform the points relative to it's transform
                    bbox = pedestrian_transform.transform_points(bbox)

                    # pedestrian's transform is relative to the world, so now,
                    # bbox contains the 3D bounding box vertices relative to the world

                    # Additionally, you can print these vertices to check that is working
                    for vertex in bbox:
                        pos_vector = np.array([
                            [vertex[0,0]],  # [[X,
                            [vertex[0,1]],  #   Y,
                            [vertex[0,2]],  #   Z,
                            [1.0]           #   1.0]]
                        ])
                        # transform the points to camera
                        transformed_3d_pos = np.dot(inv(self._extrinsic.matrix), pos_vector)
                        # transform the points to 2D
                        pos2d = np.dot(self._intrinsic, transformed_3d_pos[:3])

                        # normalize the 2D points
                        pos2d = np.array([
                            pos2d[0] / pos2d[2],
                            pos2d[1] / pos2d[2],
                            pos2d[2]
                        ])
                        # print the points in the screen
                        if pos2d[2] > 0: # if the point is in front of the camera
                            print('pos2d', pos2d)
                            x_2d = WINDOW_WIDTH - pos2d[0]
                            y_2d = WINDOW_HEIGHT - pos2d[1]
                            vertices_x, vertices_y = draw_rect(array, (y_2d, x_2d), 4, rand_color(agent.id))
```

marcgpuig, 2018-09-12:
Hi @danxuhk, does my code works for you? If it does please close the issue :)

@mhusseinsh this issue is about finding the 3D points in world space. If you have more questions, please open a new issue. The `draw_rect()` function just changes the pixel colors of the image to draw a rect on a certain position with a certain size. Those 16 points are the pixels of the image that have a new color to indicate the position of a vertx.

danxuhk, 2018-09-12:
@marcgpuig Hi Marc, thanks for your code. I tested on my side. I still have the issue. I post a new image below, and the wrong bounding boxes are annotated with black circles. The bounding boxes do not wrap the person very well. Have you observed that the 3D bounding boxes wrap the object compactly?
[image]

mhusseinsh, 2018-09-21:
@danxuhk  Hey Dan .. I tried also, and yes, sometimes I have the same problem
[image]
[image]

danxuhk, 2018-09-21:
@mhusseinsh @marcgpuig This is exactly what I have met. I do not know why this happens yet ...

mhusseinsh, 2018-09-21:
@danxuhk me neither .. and the weird thing is that it doesn't happen to all pedestrians in all frames .. so this is also something strange

marcgpuig, 2018-09-21:
Hi @danxuhk @mhusseinsh.
It is possible that you are using **Asynchronous** mode? So the image does not correspond with the data.
Try using the **Synchronous** mode. Currently, this information is in our documentation but not on the web because of the recent CARLA version change. You can find information about Synchronous vs Asynchronous mode commented in your local file [`/Docs/configuring_the_simulation.md`](https://raw.githubusercontent.com/carla-simulator/carla/master/Docs/configuring_the_simulation.md).

**Python**

```py
settings = CarlaSettings()
settings.set(SynchronousMode=True)
```
or...

**CarlaSettings.ini**

```ini
[CARLA/Server]
SynchronousMode=true
```

mhusseinsh, 2018-09-21:
@marcgpuig @danxuhk No I am using **Synchronous** mode
```py
def make_carla_settings(args):
    """Make a CarlaSettings object with the settings we need."""
    settings = CarlaSettings()
    settings.set(
        SynchronousMode=True,
        SendNonPlayerAgentsInfo=True,
        NumberOfVehicles=100,
        NumberOfPedestrians=100,
        #WeatherId=random.choice([1, 3, 7, 8, 14]),
        WeatherId=1,
        QualityLevel=args.quality_level)
```

danxuhk, 2018-09-24:
@marcgpuig I have also set the SynchronousMode as True. The weird thing is the unwrapped bounding boxes only happen on the pedestrians, not on the cars. If the setting is not correct, the issue should happen on both the cars and the pedestrians, am I right? Looking forward to your reply.

marcgpuig, 2018-09-25:
Hey @danxuhk.
From what I can see in your images, looks like that the bounding box is always extended to the direction where the pedestrian is moving. Can you confirm that?

danxuhk, 2018-09-25:
@marcgpuig Hi Marc, yes. The bounding boxes extended to the direction of the humans are moving towards.

marcgpuig, 2018-09-25:
Thanks @danxuhk. I'll take another look in a few days.

analog-cbarber, 2018-09-25:
I have seen the same problem with some pedestrians in 0.8.4.

Another strange things is that if you look at the bounding boxes at the start of the scenario when everything is spawning, the cars all drop down from above the road while the pedestrians are already on the ground, but the bounding boxes for the pedestrians also start in the air. We just ignore the initial frames in any case, but it still seems strange.

danxuhk, 2018-10-03:
@marcgpuig Hi Marc, do you have any news about this issue?:)

marcgpuig, 2018-10-05:
@danxuhk I was waiting for you to send me your code, if it possible. But anyway we are working on the new release 0.9 and we will look at this issue when we remade the bounding box system.
Sorry for the long waiting :(

danxuhk, 2018-10-11:
@marcgpuig Hi Marc, I have sent you my client code to your gmail. Could you help me to have a look? Many thanks.

danxuhk, 2019-01-21:
@marcgpuig Have you tried to have a look on this issue?

marcgpuig, 2019-01-23:
Hi @danxuhk 
Thanks for your patience. 

First of all, I had no time to take a look at the issue, but we tested all the bounding boxes using the new API where you can draw a debug box inside Unreal, and it worked. In a near future, we will add a feature to obtain all the agent transforms together with the timestamp of the frame. This will ensure the user that the image corresponds the transform.

moh3th1, 2019-03-11:
Hi @marcgpuig 

I am using the stable version of carla to draw bounding box above the lidar point cloud. I followed the same procedure you described above, but the bounding boxes still do not appear in the write place.

Here is a [snippet from the code](https://gist.github.com/moh3th1/ff9cb2375f80563e5efcadc07d86e52b) that I am using.

ThierryDubois, 2019-05-17:
Hi, @marcgpuig!

I am trying to get 2D bounding boxes to generate ground truth. Would you recommend I still use the stable version? Or has the bounding box problem not been fixed?

Harsharma2308, 2020-02-29:
@ThierryDubois  Can you tell what is this new available client with bounding boxes that you are talking about?

ThierryDubois, 2020-03-03:
> @ThierryDubois Can you tell what is this new available client with bounding boxes that you are talking about?

https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/client_bounding_boxes.py

germanros1987, 2020-03-19:
@ThierryDubois answer should solve the issue.

closed 2020-03-19 by germanros1987