carla-simulator/carla #739: Bounding box for Pedestrians

Nozomeister, 2018-08-31:
Hi CARLA team,
I have drawn the bounding boxes for pedestrian and car, but I recognized that the spawning of pedestrian with objects is random, which lead to different model of pedestrians with umbrella/trolley/bag/guitar... When drawing the bounding box using segmentation camera, I also include the trolley inside the box, which is not really correct for real life case. Is there any way to extract the attached object to pedestrian to exclude it from the bounding box or it cannot be removed as it is attached to the pedestrian when spawning?

Attached is an example of the bounding box
[image]

Thank you.

mhusseinsh, 2018-08-31:
I think in the simulator itself (even for Semantic Segmentation) .. the whole pedestrian with attached object is considered as one class

TheNihilisticRobot, 2018-09-03:
HI @Nozomeister 

We've  just checked the semantic segmentation in Carla 0.8.4 and the props our pedestrians carry are coloured different from the actual human carrying them. If that's not the case for you we should take a look at that to see what went wrong.

[image]

If you are working with the full project (not the built version) To prevent pedestrians to spawn carrying props you just need to do one change on the Walker Blueprint (Content/Blueprints/Walker), find the blueprint and inside, the "set props" function and open it, find the "Get Bool with Weight" node shown in the image below (Between "ColapsedGraph_0" node and a "Branch" node) and change its weight to 0.

[image]

closed 2018-09-14 by Nozomeister

1453042287, 2018-10-11:
@Nozomeister hi there, how did you draw the bounding box? from the segmentation camera or some way else?

Nozomeister, 2018-10-12:
> @Nozomeister hi there, how did you draw the bounding box? from the segmentation camera or some way else?

Yes, I drawed them from segmentation camera.

jagdishbhanushali, 2019-03-28:
Ho @Nozomeister , can you please share your code, how did you make it happen. I am working on 0.9.4, but I can refer your solution. If you have solved this in 0.9.4 as well, that would be great.

Thanks