carla-simulator/carla #741: [0.9.0] how to change weather

hdy117, 2018-08-31:
Hello, thank you for your brilliant work. Now I have a question
In previous version, we can change weather by editting the .ini file.
In 0.9.0 version, how do I change weather?
I also find a dynamic weather class, can I change weather while simulation is running?
If yes, how?

hdy117, 2018-09-03:
@nsubiron I think I have found the issue in Dynamic Weather. At line of 75 in file CarlaSettings.h, 

/** Load weather description from config files. (There may be overrides for each map). */
  UFUNCTION(BlueprintCallable, Category="CARLA Settings")
  void LoadWeatherDescriptions();

is defined. but maybe it is not used. so, WeatherDescriptions is empty.  

at line of  295

check(WeatherDescriptions.Num() > 0);

would throw an exception.

I called LoadWeatherDescriptions in the blueprint. 
Now it is working, I can change weather dynamically.

nsubiron, 2018-09-03:
Hi @hdy117,

The weather system is not yet implemented in the new API, that's why that function is never called. You can follow up the progress at #749.

hdy117, 2018-09-07:
@nsubiron Ok, Thanks. Looking forward to the new release of CARLA

closed 2018-11-05 by nsubiron