carla-simulator/carla #744: Can I add own solidworks 3D small car model to the Carla?

chengjinjack, 2018-09-01:
Hello.Can I add own solidworks 3D small car model to the Carla? if it can be added, how i can do the thing?

TheNihilisticRobot, 2018-09-05:
Hi @chengjinjack 

Yo might be able to import the model into unreal as long as you rig it and export it as a .FBX or .OBJ. Once you exported your model to one of this formats follow [this guide](https://carla.readthedocs.io/en/latest/how_to_add_assets/) to implement it on Carla.

chengjinjack, 2018-09-06:
@TheNihilisticRobot 
ok.Thanks!

chengjinjack, 2018-11-12:
Hello,@TheNihilisticRobot 
I have successfully implemented  own solidworks 3D small car model added  into the UE4,but I run into a issue,which is that when I press the play button,car can appear on the floor in the UE4,press w on the keyboard,car keep still and can't move forward .How can I solve this problem?