carla-simulator/carla #746: Does rosbridge work on Pre-compiled Carla

kk2491, 2018-09-01:
Hi Guys,

I am using the pre-compiled version of Carla version 0.8.2 (I have not built it from the scratch in my system).

I was trying to implement carla_ros_bridge by following the steps given in the link https://github.com/laurent-george/carla/issues/4

But I am getting the below error.

```
kk@kk-Lenovo-ideapad-320-15ISK:~/ros/catkin_ws_for_carla$ rostest carla_ros_bridge ros_bridge_client.test
[ros_bridge_client.test] is neither a launch file in package [carla_ros_bridge] nor is [carla_ros_bridge] a launch file name
kk@kk-Lenovo-ideapad-320-15ISK:~/ros/catkin_ws_for_carla$ 
```

Does this method works on the pre-compiled Carla or do I have to build Carla from github code ?

Thank you,
KK

laurent-george, 2018-09-06:
Currently ros-bridge supports only Carla python API from 0.8.* version, thus should work with 0.8.2. But your problem does not seems to be related to CARLA version, it is more a ros related issue, seems like carla_ros_bridge is not found in your ros workspace.

 have you source your ros setup script in the same shell that the one you are using to run rostest command ?

   source ~/ros/catkin_ws_for_carla/devel/setup.bash

kk2491, 2018-09-08:
@laurent-george Thanks for the response. 

I am able to do it now using pre-compiled package version 0.8.2 and copying the carla_ros_bridge from the github master code. 

However I am facing a new error now. Will be posting the error soon.

Thank you,
KK

closed 2018-09-10 by laurent-george