carla-simulator/carla #750: Plugin system for adding new sensors

nsubiron, 2018-09-03:
Although due to performance reasons is not feasible to have a runtime plugin system for sensors, we can do it at compile time.

Make a system in which users can add their own sensors by

1) Adding a sensor class to UE4 that gathers data.
2) Adding a serializer class to LibCarla with two functions, `serialize` and `deserialize`.
3) Registering both classes at some sensor registry.

nsubiron, 2018-10-02:
@analog-cbarber I did the "ShaderBasedSensor" that you just need to provide a post-processing material as you suggested :)

We also make some testing for changing the format of the image buffer but gave us some issues, most of the formats doesn't seem to be supported by UE4 captures (or probably we are doing something wrong) so for now I'm not adding that until we understand well how to do it.

marcgpuig, 2018-10-03:
Hey @analog-cbarber
[Here](https://api.unrealengine.com/INT/API/Runtime/Core/EPixelFormat/index.html) you can find the full list of Unreal formats, but sadly as @nsubiron said, the render target supports only these formats:
```
PF_B8G8R8A8
PF_A16B16G16R16
PF_FloatRGB
PF_FloatRGBA
PF_A2B10G10R10
PF_DepthStencil
```
It could be super cool support something like `PF_A32B32G32R32F` to return a buffer with world position per pixel, or another one with `PF_R32_FLOAT` to simply return the depth without codification.

Not sure the amount of work that should take changing the engine to support these features, but we are not really interested in it due to the costs of maintaining a custom Unreal.

PR #830 (2018-10-23): New sensor system that aims to make easier adding new sensors
#### Description

The idea is to create a new sensor system that allows contributors to "plug in" their implementations easily.

  1. Add a sensor actor.
  2. Add a sensor serializer (and possibly a sensor data view in Python).
  3. Register the classes.

All the details below.

This PR fixes:

  - Fixes #750 Plugin system for adding new sensors.
  - Fixes #687 Add Lidar to new API.
  - I have also removed all the legacy sensor code.

**TODO:**

- [x] Test that the sensor transform received is in world coordinates.
- [ ] Test the re [... cut]
closed 2018-10-23 by nsubiron