carla-simulator/carla #753: Issue in configuring ros_bridge

kk2491, 2018-09-03:
Hi Guys,

This  issue has been reported before, however the approach I used is different so I am posting it as a new issue.

System : Ubuntu 16.04
Carla Version : 0.8.4 - Pre-compiled (downloaded from https://drive.google.com/file/d/18OaDbQ2K9Dcs25d-nIxpw3GPRHhG1r_2/view)

I have built the Carla from the source code (using the master branch from Github) however I was not able to find the CarlaUE4.sh file to launch the simulator. So I have used the below approach to configure the ros_bridge.

1. Downloaded the pre-compiled Carla (0.8.4)
2. Found there is no carla_ros_bridge in the directory. 
3. Copied the carla_ros_bridge from built code to the pre-compiled one
4. Followed the procedure given in the below link <br /> https://github.com/carla-simulator/carla/tree/master/carla_ros_bridge
5. While running the below command
```rostest carla_ros_bridge ros_bridge_client.test``` 
Got the error : ```[ros_bridge_client.test] is neither a launch file in package [carla_ros_bridge] nor is [carla_ros_bridge] a launch file name```
6. Fixed it by sourcing the catkin workspace <br />     ```source ~/ros/catkin_ws_for_carla/devel/setup.bash ```
7.  Now I am getting the below error, could you please help me to get the solution for this please.
```kk@kk-Lenovo-ideapad-320-15ISK:~$ rostest carla_ros_bridge ros_bridge_client.test
... logging to /home/kk/.ros/log/rostest-kk-Lenovo-ideapad-320-15ISK-7012.log
[ROSUNIT] Outputting test results to /home/kk/.ros/test_results/carla_ros_bridge/rostest-test_ros_bridge_client.xml
Traceback (most recent call last):
  File "/home/kk/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/client.py", line 8, in <module>
    from carla.client import make_carla_client
  File "/home/kk/Carla/CARLA_0.8.4/PythonClient/carla/client.py", line 14, in <module>
    from . import sensor
  File "/home/kk/Carla/CARLA_0.8.4/PythonClient/carla/sensor.py", line 19, in <module>
    from .transform import Transform, Translation, Rotation, Scale
  File "/home/kk/Carla/CARLA_0.8.4/PythonClient/carla/transform.py", line 18, in <module>
    from . import carla_server_pb2 as carla_protocol
  File "/home/kk/Carla/CARLA_0.8.4/PythonClient/carla/carla_server_pb2.py", line 6, in <module>
    from google.protobuf import descriptor as _descriptor
  File "/home/kk/.local/lib/python2.7/site-packages/google/protobuf/__init__.py", line 37, in <module>
    __import__('pkg_resources').declare_namespace(__name__)
  File "/home/kk/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 55, in <module>
    from pkg_resources.extern import six
  File "/home/kk/.local/lib/python2.7/site-packages/pkg_resources/extern/__init__.py", line 43, in load_module
    __import__(extant)
  File "/home/kk/.local/lib/python2.7/site-packages/pkg_resources/_vendor/six.py", line 1, in <module>
    """Utilities for writing code that runs on Python 2 and 3"""
KeyboardInterrupt
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/rostest/nodes/publishtest", line 155, in <module>
    rostest.run(PKG, NAME, PublishTest, sys.argv)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/__init__.py", line 141, in rosrun
    suite = unittest.TestLoader().loadTestsFromTestCase(test)
  File "/usr/lib/python2.7/unittest/loader.py", line 56, in loadTestsFromTestCase
    loaded_suite = self.suiteClass(map(testCaseClass, testCaseNames))
  File "/opt/ros/kinetic/share/rostest/nodes/publishtest", line 94, in __init__
    rospy.init_node(NAME)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    raise rospy.exceptions.ROSInitException("init_node interrupted before it could complete")
rospy.exceptions.ROSInitException: init_node interrupted before it could complete
[Testcase: testtestTopics] ... ERROR!
ERROR: max time [20.0s] allotted for test [testTopics] of type [rostest/publishtest]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

rostest log file is in /home/kk/.ros/log/rostest-kk-Lenovo-ideapad-320-15ISK-7012.log
```

Is the approach I used is feasible ?

In case if we build the project from Scratch I dont see the CarlaUE4.sh in the built directory. Could you please clarify?

Thank you,
KK

marcgpuig, 2018-09-04:
Hi @kk2491.
> I have built the Carla from the source code (using the master branch from Github) however I was not able to find the CarlaUE4.sh file to launch the simulator.

First of all, take a look at well-known issue #22 ;)

closed 2018-09-10 by laurent-george

xuwenqian66, 2018-12-11:
> Hi Guys,
> 
> This issue has been reported before, however the approach I used is different so I am posting it as a new issue.
> 
> System : Ubuntu 16.04
> Carla Version : 0.8.4 - Pre-compiled (downloaded from https://drive.google.com/file/d/18OaDbQ2K9Dcs25d-nIxpw3GPRHhG1r_2/view)
> 
> I have built the Carla from the source code (using the master branch from Github) however I was not able to find the CarlaUE4.sh file to launch the simulator. So I have used the below approach to configure the ros_bridge.
> 
> 1. Downloaded the pre-compiled Carla (0.8.4)
> 2. Found there is no carla_ros_bridge in the directory.
> 3. Copied the carla_ros_bridge from built code to the pre-compiled one
> 4. Followed the procedure given in the below link  https://github.com/carla-simulator/carla/tree/master/carla_ros_bridge
> 5. While running the below command
>    `rostest carla_ros_bridge ros_bridge_client.test`
>    Got the error : `[ros_bridge_client.test] is neither a launch file in package [carla_ros_bridge] nor is [carla_ros_bridge] a launch file name`
> 6. Fixed it by sourcing the catkin workspace      `source ~/ros/catkin_ws_for_carla/devel/setup.bash `
> 7. Now I am getting the below error, could you please help me to get the solution for this please.
> 
> ```
> ... logging to /home/kk/.ros/log/rostest-kk-Lenovo-ideapad-320-15ISK-7012.log
> [ROSUNIT] Outputting test results to /home/kk/.ros/test_results/carla_ros_bridge/rostest-test_ros_bridge_client.xml
> Traceback (most recent call last):
>   File "/home/kk/ros/catkin_ws_for_carla/src/carla_ros_bridge/src/carla_ros_bridge/client.py", line 8, in <module>
>     from carla.client import make_carla_client
>   File "/home/kk/Carla/CARLA_0.8.4/PythonClient/carla/client.py", line 14, in <module>
>     from . import sensor
>   File "/home/kk/Carla/CARLA_0.8.4/PythonClient/carla/sensor.py", line 19, in <module>
>     from .transform import Transform, Translation, Rotation, Scale
>   File "/home/kk/Carla/CARLA_0.8.4/PythonClient/carla/transform.py", line 18, in <module>
>     from . import carla_server_pb2 as carla_protocol
>   File "/home/kk/Carla/CARLA_0.8.4/PythonClient/carla/carla_server_pb2.py", line 6, in <module>
>     from google.protobuf import descriptor as _descriptor
>   File "/home/kk/.local/lib/python2.7/site-packages/google/protobuf/__init__.py", line 37, in <module>
>     __import__('pkg_resources').declare_namespace(__name__)
>   File "/home/kk/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 55, in <module>
>     from pkg_resources.extern import six
>   File "/home/kk/.local/lib/python2.7/site-packages/pkg_resources/extern/__init__.py", line 43, in load_module
>     __import__(extant)
>   File "/home/kk/.local/lib/python2.7/site-packages/pkg_resources/_vendor/six.py", line 1, in <module>
>     """Utilities for writing code that runs on Python 2 and 3"""
> KeyboardInterrupt
> Traceback (most recent call last):
>   File "/opt/ros/kinetic/share/rostest/nodes/publishtest", line 155, in <module>
>     rostest.run(PKG, NAME, PublishTest, sys.argv)
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/__init__.py", line 141, in rosrun
>     suite = unittest.TestLoader().loadTestsFromTestCase(test)
>   File "/usr/lib/python2.7/unittest/loader.py", line 56, in loadTestsFromTestCase
>     loaded_suite = self.suiteClass(map(testCaseClass, testCaseNames))
>   File "/opt/ros/kinetic/share/rostest/nodes/publishtest", line 94, in __init__
>     rospy.init_node(NAME)
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
>     raise rospy.exceptions.ROSInitException("init_node interrupted before it could complete")
> rospy.exceptions.ROSInitException: init_node interrupted before it could complete
> [Testcase: testtestTopics] ... ERROR!
> ERROR: max time [20.0s] allotted for test [testTopics] of type [rostest/publishtest]
>   File "/usr/lib/python2.7/unittest/case.py", line 329, in run
>     testMethod()
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
>     self.test_parent.run_test(test)
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
>     return self.runner.run_test(test)
>   File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
>     (test.time_limit, test.test_name, test.package, test.type))
> --------------------------------------------------------------------------------
> 
> [ROSTEST]-----------------------------------------------------------------------
> 
> 
> SUMMARY
>  * RESULT: FAIL
>  * TESTS: 0
>  * ERRORS: 1
>  * FAILURES: 0
> 
> rostest log file is in /home/kk/.ros/log/rostest-kk-Lenovo-ideapad-320-15ISK-7012.log
> ```
> Is the approach I used is feasible ?
> 
> In case if we build the project from Scratch I dont see the CarlaUE4.sh in the built directory. Could you please clarify?
> 
> Thank you,
> KK

hi, have you already fixed this problem? I compiled the ros_bridge.py without any problem, but it still doesn't work. Do you know what may happens?