carla-simulator/carla #755: Speed Segmentation - Can we label the actor's velocity rather than its label?

lshafir, 2018-09-04:
We have a point cloud simulation using Carla.
We are trying to produce ground truth values per each point in the cloud whether it's moving or not.

In the past we used the collision engine (ShootLaser) of the server (with our custom directions map) to produce the point cloud, but due to limitations it has for moving objects, we are currently using the depth map to transform the 2D image into 3D point cloud.
We also use the 'SemanticSeg' post processing to have a type label per each pixel (point in the cloud).

Our first approach was to to have an instance segmentation per point rather than type segmentation, but we saw that the current buffers available to mark the object ID are only 0-255.
Then we tried to put a boolean value (as a stencil value) based on the Actor velocity (using GetVelocity())
but it appears that in Tagger.cpp the velocity of all the actors is always zero.

Any advice about generating speed segmentation will be appreciated :)

Thanks!

marcgpuig, 2018-09-04:
Hello @lshafir.

> but we saw that the current buffers available to mark the object ID are only 0-255

That's our problem too!
To make clear, you are interested **only** in knowing if the object is moving or not?
The way I would do it would be checking the car transformation, and see if it is different from the previous frame. If this is the case, you can iterate all the points in your point cloud relevant to this car and set the information (like you said). How to find these points? Check out this issue in case you don't have done it #76. This is all we know on this topic.
We have plans to do this instance semantic segmentation in the future.

lshafir, 2018-09-04:
Hello @marcgpuig 

Thanks! Indeed, we checked out #76, but finding all the points in the point cloud that relevant to an object is not accurate in case we rely on the bounding boxes.

Do you know why in `Tagger.cpp` (where the labels are marked) the speed of the actor is always zero? Is there any other way to retrieve the velocity under this context (while iterating the static and skeletal meshes)?

closed 2018-11-12 by stale[bot]

strohhut22, 2022-06-29:
Thanks for your work on the Carla simulator and the instance segmentation that has been implemented is great!
My question: Is there a really straightforward way to get annotations of moving or static actors in the current version of Carla (0.9.13)? Comparing the instance centers of objects to the previous frame works like a charm, however, some minor false positives occur when points appear or disappear from a stationary object due to, e.g., occlusion, because then the bounding box center and mean of the instance's points change.