carla-simulator/carla #758: Simulation for UAV(s)

whxru, 2018-09-05:
Is it possible to append UAV/drone/multirotor as a supported (or even default) vehicle and which can coexist with cars ?

nsubiron, 2018-09-06:
Hi @whxru,

This is not something we plan to add any time soon, our priority is simulating wheeled vehicles so all our efforts go in that direction. For that you have simulators like AirSim. 

It would be however a great community contribution :wink:

analog-cbarber, 2018-09-06:
The controls and physics for flying vehicles would be entirely different. AirSim is probably a better choice if you care about control scenarios.

If you just want a magic flying sensor platform for generating data for machine learning
it probably would not be that hard to support a flying pawn that required the user to set
its position at each frame or provide a simple "move to position x at speed s" type api.

whxru, 2018-09-07:
@nsubiron Thanks for your reply, I'm actually using AirSim now, but AirSim do not allow multirotors coexist with cars. I think new features of CARLA in version 0.9.0 is really cool to do researches in field of multi-agent and I'm doing some work on network of UAV-UGV now, although simulation for UAV(s) will not be supported, but using a flying camera could be really helpful for my current work, thanks again.

closed 2018-09-07 by whxru