carla-simulator/carla #760: White screen when running Carla version 0.9.0

mselli, 2018-09-05:
Hi! 
 I tried running the compiled version of CARLA 0.9.0 downloaded from github and all I see is a white screen for the server. I am using carla 0.9.0 on Ubuntu 16.04 on a i7 with a gpu Nvidia gtx1060.
I can run the example.py script successfully, although the server remains with a white screen like in the picture. 
The same thing happens when I use the packaged version that I created when I do make package  from the master code pulled from the repo. 

Do you know what could be the problem?

[image]

nsubiron, 2018-09-06:
Hi @mselli,

I've never seen anything like this, can it be you are on an old computer? https://answers.unrealengine.com/questions/772557/whenever-i-try-to-open-unreal-4190-it-just-opens-a.html

Take also a look at #757, may also be a problem with the GPU drivers.

nsubiron, 2018-09-06:
Oh see this one too https://devtalk.nvidia.com/default/topic/1036275/unreal-engine-4-games-showing-white-screen-with-396-24-02-drivers/#5264166

mselli, 2018-09-10:
Thanks for all the links! 
For some reason that I don't really know, the view mode "Lit" from Unreal Engine  doesn't properly work on my PC (gives a white screen) and I couldn't change it to any other view mode (Unlit, Wireframe) because the debug view modes are disabled (unless I am using the Unreal editor). So I just added:

`r.ForceDebugViewModes=1`
to the DefaultEngine.ini under the [RenderSettings] on the mastar carla project and packaged it. 

Now when I run the CarlaUE4.sh script I can switch to the "unlit" mode and works fine.
I'll use this solution while I find the real reason why the Lit mode doesn't work on my PC, but that's not a CARLA issue. 
Thanks!

closed 2018-09-10 by mselli

pleasenott, 2024-06-06:
Hello, I run ./CarlaUE4.sh but just get this.
```
GameThread timed out waiting for RenderThread after 60.00 secs
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=139728 LargeMemoryPoolOffset=270864 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```
and I find that your UE4 report so much?
which parameters can help me get more message .