carla-simulator/carla #761: Black screen in manual_control.py in Windows

nsubiron, 2018-09-06:
Version 0.9.0.

The actual problem is that the connection to the streaming port fails (and the failure message is debug only). Somehow using the tcp endpoint as given only works on Linux. I've tried using a resolver instead and seems to work.

```cpp
boost::asio::ip::tcp::resolver resolver(_socket.get_io_service());
auto endpoint_it = resolver.resolve({"127.0.0.1", "2001"});
boost::asio::async_connect(_socket, endpoint_it, _strand.wrap(handle_connect));
```

M-Hammod, 2018-09-06:
could you please clarify where in manual_control.py I should add this code, there is no reference to the streeming port 2001 in that file, I see only 2000

nsubiron, 2018-09-06:
Hi @M-Hammod, that's C++ part of the LibCarla library, I opened a PR to fix this, it's now waiting review. It hasn't been tested in Windows yet.

commit 1f1f24da (2018-09-06): Fix #761 streaming client fails to connect in Windows