carla-simulator/carla #764: Changing the player vehicle

link-er, 2018-09-07:
Hello!

I want to record separate distinct datasets of camera images by running autopilot and I would prefer to have distinct cars as player vehicle. I am changing the setting PlayerVehicle as it is documented to various values following https://github.com/carla-simulator/carla/issues/671 , but I always get the same default Mustang.
I tried it both with Carla0.8.2 and Carla 0.8.4. I am running the prebuild version for Ubuntu with 

`./CarlaUE4.sh /Game/Maps/Town01 -benchmark -fps=25 -windowed -ResX=1000 -ResY=500 -carla-server`

and

`python client_example.py --autopilot -i`

I am changing the value of PlayerVehicle inside of the script client_example.py and even (maybe it is used for the server?) in the Example.CarlaSettings.ini, but nothing affects the car

Can you suggest what is the problem?

link-er, 2018-09-28:
Hi @marcgpuig.
Thank you for the answer - providing the settings file when starting server did change the car finally. But I am a bit confused - I thought that settings of the client overlay settings of the server? If no, then does it mean that all the settings would be always taken from the server settings? The weather is changing when I change it in the client script though.
And now, when I changed the car, in some models I can see the frontal window in the recorded data. I thought I can get read of it by moving the camera (camera.set_position(x,y,z)), but I tried to change it both in server settings and in client settings without any success. Any ideas here?
[image]

link-er, 2018-10-01:
@marcgpuig I am using PythonClient example code basically, with slight changes of initial positions. But I definitely first start server and then Client - so that is the reason then? I do not have any uniting code, I am just starting both scripts manually in separate terminals.

link-er, 2018-10-01:
@marcgpuig  I tried it with the position you set in both settings - server and client - and I still has the same image. 
Here is my client code
```
from __future__ import print_function

import argparse
import logging
import random
import time

from carla.client import make_carla_client
from carla.sensor import Camera, Lidar
from carla.settings import CarlaSettings
from carla.tcp import TCPConnectionError
from carla.util import print_over_same_line

TRAIN_POSITIONS_T1 = [  0,   1,   2,   3,   4,   5,   6,   7,   8,   9,  10,  11,  12,
        13,  14,  21,  26,  27,  28,  29,  30,  31,  32,  33,  34,  43,
        54,  65,  72,  73,  78,  79,  80,  81,  82,  83,  84,  85,  86,
        87,  88,  89,  94,  95,  96,  97,  98,  99, 100, 101, 102, 103,
       104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116,
       117, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130,
       131, 132, 133, 140, 149, 150, 151, 118]

def run_carla_client(args):
    # Here we will run train episodes with 400 frames each.
    number_of_episodes = len(TRAIN_POSITIONS_T1)
    frames_per_episode = 400

    # We assume the CARLA server is already waiting for a client to connect at
    # host:port. To create a connection we can use the `make_carla_client`
    # context manager, it creates a CARLA client object and starts the
    # connection. It will throw an exception if something goes wrong. The
    # context manager makes sure the connection is always cleaned up on exit.
    with make_carla_client(args.host, args.port) as client:
        print('CarlaClient connected')
        driving_commands = open("driving_commands.txt", "w")
        coordinates = open("coordinates.txt", "w")
        speeds = open("speeds.txt", "w")

        for episode in range(0, number_of_episodes):
            # Start a new episode.

            # Create a CarlaSettings object. This object is a wrapper around
            # the CarlaSettings.ini file. Here we set the configuration we
            # want for the new episode.
            settings = CarlaSettings()
            settings.set(
                SynchronousMode=True,
                SendNonPlayerAgentsInfo=True,
                NumberOfVehicles=20,
                NumberOfPedestrians=40,
                WeatherId=random.choice(range(15)),
                QualityLevel=args.quality_level)
            settings.randomize_seeds()

            # Now we want to add a couple of cameras to the player vehicle.
            # We will collect the images produced by these cameras every
            # frame.
    
            # The default camera captures RGB images of the scene.
            camera0 = Camera('CameraRGB')
            # Set image resolution in pixels.
            camera0.set_image_size(800, 600)
            # Set its position relative to the car in meters.
            camera0.set_position(2.0, 0.0, 1.4)
            settings.add_sensor(camera0)

            # Now we load these settings into the server. The server replies
            # with a scene description containing the available start spots for
[... 245 lines cut ...]
[CARLA/Sensor/MyLidar]
SensorType=LIDAR_RAY_CAST
; Number of lasers.
Channels=32
; Measure distance in meters.
Range=50.0
; Points generated by all lasers per second.
PointsPerSecond=100000
; Lidar rotation frequency.
RotationFrequency=10
; Upper and lower laser angles, positive values means above horizontal line.
UpperFOVLimit=10
LowerFOVLimit=-30
; Position and rotation relative to the vehicle.
PositionX=0
PositionY=0
PositionZ=1.40
RotationPitch=0
RotationYaw=0
RotationRoll=0
```
And I start server this way:
`./CarlaUE4.sh /Game/Maps/Town01 -benchmark -fps=25 -windowed -ResX=1000 -ResY=500 -carla-server -carla-settings="Example.CarlaSettings.ini"`
And client this way:
`python training_dataset_creator.py --autopilot -i`

marcgpuig, 2018-10-02:
@link-er 
Well, this is weird, your code looks good to me...
Can you try something like `camera0.set_position(2.0, 0.0, 100.0)` to exaggerate the camera position and share your results?

link-er, 2018-10-04:
@marcgpuig 
With the height 100 it showed completely out of scope landscape =)
But it helped me to understand what is the problem - I was not restarting client script, only the server, after trying to adjust camera. Thus settings were not overridden and the image of the previous setup was displayed.
Thank you for help!

closed 2018-10-04 by link-er