carla-simulator/carla #768: connection error manual_control

mhusseinsh, 2018-09-10:
whenever I run the manual_control .. I receive the following error
```
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server localhost:2000

Welcome to CARLA manual control.

Use ARROWS or WASD keys for control.

    W            : throttle
    S            : brake
    AD           : steer
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot

    R            : restart level

STARTING in a moment...

Starting new episode...
Step 1 (0.4 FPS): 0.00 km/h, 0% other lane, 0% off-road                                                                 ERROR: (localhost:2002) failed to write data: [WinError 10054] An existing connection was forcibly closed by the remote host
ERROR: (localhost:2000) failed to read data: [WinError 10054] An existing connection was forcibly closed by the remote host
Starting new episode...
Step 1 (0.5 FPS): 0.00 km/h, 0% other lane, 0% off-road                                                                 ERROR: (localhost:2002) failed to write data: [WinError 10054] An existing connection was forcibly closed by the remote host
ERROR: (localhost:2000) failed to read data: [WinError 10054] An existing connection was forcibly closed by the remote host
```
However, when i run the client_example, it runs fine

mhusseinsh, 2018-09-24:
@nsubiron hello
any idea about this issue ? I always face the same problem
it runs a little, then the connection is closed
STARTING in a moment...

```
Starting new episode...
Step 42 (0.4 FPS): 24.37 km/h, 0% other lane, 0% off-road            ERROR: (localhost:2001) failed to read data: [WinError 10054] An existing connection was forcibly closed by the remote host
ERROR: (localhost:2000) failed to read data: [WinError 10054] An existing connection was forcibly closed by the remote host
Starting new episode...
Step 219 (0.3 FPS): 5.64 km/h, 0% other lane, 0% off-road            Starting new episode...
Step 205 (0.5 FPS): -0.00 km/h, 0% other lane, 0% off-road           ERROR: (localhost:2001) failed to read data: [WinError 10054] An existing connection was forcibly closed by the remote host
ERROR: (localhost:2000) failed to read data: [WinError 10054] An existing connection was forcibly closed by the remote host
Starting new episode...
Step 108 (0.3 FPS): 19.68 km/h, 0% other lane, 0% off-road 

```

nsubiron, 2018-09-24:
Hi @mhusseinsh,

Make sure to use the same version of the client and the simulator and only one simulator running in the same port. Check also that no other application is using that port.

mhusseinsh, 2018-09-24:
how to make sure of this ?

nsubiron, 2018-09-24:
Use the client and simulator that come together in the same release package. Checking the port depends on your OS, but you can start by testing different ports.

mhusseinsh, 2018-09-25:
@nsubiron they are from the same release package yes, v0.8.4
I am running on Windows 10

closed 2018-12-01 by stale[bot]

josiahcoad, 2020-06-29:
I am having this issue with carla v0.8.2. PythonClient/manual_control.py and PythonClient/example_client.py run ok but reinforcement_learning/run_RL.py and PythonClient/driving_benchmark_example.py result in this error :(

fjzs, 2023-07-18:
is there any formal solution to this issue?