carla-simulator/carla #770: Ray-cast based Lidar labels

sanketgujar, 2018-09-11:
Great work implementing Lidar in Carla. I wanted to know how can I get the ground-truth labels (3D bounding boxes/ Pointwise labels) for the Lidar data for the objects in the scene.

analog-cbarber, 2018-09-11:
In 0.8 you can get bounding box information for all vehicles and pedestrians. You can use that to find out whether a lidar point belongs to one of those actors (more or less). 

For some limited camera view, you could get CARLA to generate 2D semantic segmentation information and project the lidar points on to the view plane to get semantic labels.

If those techniques are not good enough, you would probably have to make your own version of the Lidar sensor.

sanketgujar, 2018-09-11:
Thanks for the quick reply, It will be a great help if you can also answer the following question
1. Is the bounding boxes will be in Lidar coordinates?
2. Where can I get the transformations for the projections?

marcgpuig, 2018-09-12:
Hi @sanketgujar:

> Is the bounding boxes will be in Lidar coordinates?

Take a look here #392.

> Where can I get the transformations for the projections?

And this can help you https://github.com/carla-simulator/carla/issues/293#issuecomment-376528581 :)

sanketgujar, 2018-09-15:
The projection of 2D semantic segmentation information worked for me. Thanks a lot !!!

closed 2018-09-15 by sanketgujar

vishal-prgmr, 2019-05-07:
@sanketgujar , I think its possible to directly get the pointwise labels. Still looking for it?