carla-simulator/carla #773: Link error in step 'make package'

GloryWoo, 2018-09-13:
I've downloaded carla and UnrealEngine_4.19, when I make this project as instructions, I meet a problem.
'make launch'  success, and there is UnrealEngine displaying.  When 'make package' , compile success, but there are link errors:
[UnrealBuildTool-2018.09.13-11.01.45.txt](https://github.com/carla-simulator/carla/files/2377558/UnrealBuildTool-2018.09.13-11.01.45.txt)

Performing 1 actions (4 in parallel)
  [1/1] Link (lld) CarlaUE4
  /usr/lib/llvm-5.0/bin/ld.lld: error: undefined symbol: ACityMapGenerator::ACityMapGenerator(FObjectInitializer const&)
>>> referenced by CarlaMapGenerator.cpp:11

How could I solve this problem?

nsubiron, 2018-09-13:
That's odd, those are all functions present in the Carla plugin, somehow did not compile correctly? Have you modified anything in the configuration?

And just to be sure, try doing a clean package, Unreal caches files and sometimes fails with weird things because it's caching something from a previous version.

    make clean package

GloryWoo, 2018-09-14:
After make clean and make package in absolute path, the problem is solved.
(I use 'ln -s' to link slave disc to main disk and project is in slave disc, maybe result in this problem)

closed 2018-09-14 by GloryWoo