carla-simulator/carla #774: Issue on creating my own map

quanheng123, 2018-09-13:
Hi, I have a issue on creating my own map, I choose a default map and delete everything I don't need just as the "Map  customization" describes, and I add a new road in the default Map also the NviModifierVolumen , RoadNavmesh and Routplanner in every intersection, but somehow, when I klick Trigger Road Map generation in CalraMapGenerator, my new-add-road can not be fully recognized(No green arrows in my new adding road), can somebody please tell me what I missed?  
Thanks.

TheNihilisticRobot, 2018-09-13:
Hi @quanheng123 

I'm not very sure what your pipeline is like. Did you manually created a road and then tried to use CarlaMapGenerator to define the road directions? Trigger Road Map Generation only works with roads created by a CarlaMapGenerator. For manually added ones you are stuck with the RoutePlanners. Those direct the vehicles however you please but won't give lane information to the controller like the standard roads do, hence the lack of green arrows.

quanheng123, 2018-09-13:
> Hi @quanheng123
>
> I'm not very sure what your pipeline is like. Did you manually created a road and then tried to use CarlaMapGenerator to define the road directions? Trigger Road Map Generation only [...]

Hello, @TheNihilisticRobot  
Thanks for the answer! 
I think I did create the road with CarlaMapGenerator  Here are the details which I create my own road : 

1: I use the Static Meshes which in Content file(eg:St_Road_TCross_Road0 to Road9 and the landmarking to build a T Cross intersection), then I attached the whole static Meshes from New Road to CarlaMapGenerator (then these Static meshes are now in the CarlaMapgenerator. So Is this what you mention about: created by CarlaMapGenerator ???)

2. I added it in to the default Map and try to adjust it coordinate , try to make it combination the road in the default Map ,but before I have deleted  everything I don't need, only the Static Meshes in CarlaMapGenerator , Lighting , PostProcessVolumen  and the RoadNavMesh NavMeshBoundsVolum(I dont know what are these three volume's' functions , so I haven't deleted  ) also the NavMeshVolumen in every intersections and Playerstarts.
3. So after check , I click the Trigger Road Map Generation, It shows that it still can't recognize all of my  Road.
So. What did I missed? 
Thanks again for the answers!

TheNihilisticRobot, 2018-09-13:
So, that is not the expected use for the road map generator.

Starting With an empty level, place a CarlaMapGenerator in the level. Once placed, under "Map Generation" you'll have 6 parameters that combined will generate a road for you. Once size and seed are defined, checking "Generate Roads" will give you a city layout. These layout is what that will be used when triggering the road map generation. You may keep changing the seed until you are pleased with the current layout. But you can't alter it manually. If you add extra roads using external assets you might direct traffic around it using the route planners but those roads won't have the lane information our standard roads have.

quanheng123, 2018-09-13:
Thanks.@TheNihilisticRobot  
So here I have other two questions : 
1. Using the Routeplanner with the lack of lane information will influence the Auto Navigation of the Automobiles?
2. How can I modify the Static Mesh (eg:  change the width and length of the Static mesh : St_road_Curve_Road0)?

quanheng123, 2018-09-13:
And also can you pls show me where I can find  the tutorials of the Routeplanner? @TheNihilisticRobot    Thanks.

quanheng123, 2018-09-13:
And also I have the question 3 about the Route planner, when I add a Route planner in maps , How can I added more points in one directions(more than 2 or 3 in the intersection) to design the trajectory of the line of the Route planner.
Thanks.

TheNihilisticRobot, 2018-09-14:
1: The RoutePlaner alone is able to direct our AI vehicles without a problem, the downside is that you won't have information about your position on the lane. You'll have to cover every road you made with Routeplanners, though. So some extra effort is required.

2: Though you can't Modify the meshes as part of the CarlaMapGenerator the model itself is just a model and is treated as such by the engine. You might move, rotate or scale it like any other. Scale tool is binded to the "R" Key. See [Epic's Level Designer Quick Start Guide](https://docs.unrealengine.com/en-US/Engine/QuickStart) for more information.

3: The component that defines the different routes out of a RoutePlanner is a Spline and works like such (See: [How-To Edit a Spline Component in the Viewport](https://docs.unrealengine.com/en-US/Engine/BlueprintSplines/HowTo/EditSplineComponentInEditor)). For each route you want to create you'll need to find In the RoutePlanner's details panel, under "Traffic Routes" the "Routes" array and add an element. Each element of this array is a route and will have its corespondent spline and a probability variablethat defines the probability of being chosen by any vehicle going through the RoutePlanner.

quanheng123, 2018-09-14:
Thanks for the patient answers! @TheNihilisticRobot  
 But about the first answer, what is the specific meaning of my position?  Is it the coordinate parameter (location rotation something) in road level (not the world coordination)?

TheNihilisticRobot, 2018-09-17:
Every vehicle on Carla "Knows" what percentage of itself is out of the road or into the oposite lane. That is the information you'll be missing if you use RoutePlaners and not the CarlaMapGenerator roads.

quanheng123, 2018-09-17:
Got it,Thanks for the help,appreciated !

closed 2018-09-17 by quanheng123