carla-simulator/carla #775: Is it possible to run client on a remote server?

aquiire, 2018-09-13:


FarzanehJafari, 2018-09-15:
Hi, I want run carla server on a computer and client on an another computer. actully, I want run clinet server on two computers seperated together. I dont found any way for doing it. please help me

FarzanehJafari, 2018-09-17:
> Carla is network based, it should be possible to just specifiy the server adress in your client. Have you tried that?

Thanks a lot for your helpI tried to connect them again yesterday and it was a success. I set servers ip for client and server with this command (—host ip adrress).

aquiire, 2018-09-18:
@grafandreas Can we do this over SSH? (I'm a newbie) I can run jupyter via ssh. Would just SSH tunneling   the ports 2000, 2001, 2002 be enough?

micmarty, 2018-12-08:
@aquiire 
```
Carla Server: Ubuntu 16.04 Desktop (my local machine -> carla renderer)
Carla Client (PythonAPI): Ubuntu 16.04 Server (remote cluster -> model trainings)
```
Remote port forwarding works for me without any issues.
You must check if your remote machine allows for remote port forwarding (AFAIK check it out somewhere inside `/etc/ssh/sshd_config` -> if it's not yours you'll have to ask administrator)

My local machine config (`~/.ssh/config`)
```

Host SOME_HOSTNAME_ALIAS
    HostName SOME_HOSTNAME
    User MY_REMOTE_USERNAME
    RemoteForward 2000 127.0.0.1:2000
    RemoteForward 2001 127.0.0.1:2001
    RemoteForward 2002 127.0.0.1:2002
```