carla-simulator/carla #777: Obtaining high level commands

SoumiDas, 2018-09-13:
Hello,
Can you tell me how do I obtain high level commands by running CARLA In autopilot mode? It shall be of great help!

Thanks!

marcgpuig, 2018-09-14:
Hi @SoumiDas 
Which kind of high-level commands do you mean?

SoumiDas, 2018-09-14:
I mean the 4 high level commands 'Straight' , 'Left', 'Right' ,'Follow Lane' which are one of the target values in the creation of CARLA dataset.

srijanreddy98, 2019-03-10:
Can the gps navigation features be added like: 
- Take the next left/right
- Take the third exit at this circle/roundabout

MalekWahidi, 2024-01-01:
> I mean the 4 high level commands 'Straight' , 'Left', 'Right' ,'Follow Lane' which are one of the target values in the creation of CARLA dataset.

Same issue here, anyone figured this out?