carla-simulator/carla #779: GPU-accelerated LIDAR based on z-buffer

marcgpuig, 2018-09-14:
### A new sensor!
The plan is to create a new LIDAR based on the depth render using cameras.

### Why?
- This sensor will generate **geometry perfect points**, unlike the current one that returns points based on the collision box for animated objects (like cars and pedestrians).
- **Improved information:** the idea is that you will be able to get the 3D points alongside with the information different render buffers like semantic segmentation (lidar now detecting road lines yay!).
- **More information types:** thanks to the previous point, now we will be able to get more information like per point: RGB, world normal, roughness, metallic information and so on.

johnzjq, 2018-12-25:
The present Lidar is too slow for online simulation, 4fps in a 2700x 8 core cpu. Rending four images and extracting depth from them are not that difficult, but they are crucial for online simulation tasks.

bigsheep2012, 2019-02-01:
Hello @marcgpuig and Carla team,
Thanks for releasing 0.9.3.
Just want to confirm, according to the change log, the new lidar (for generating perfect geometry point) is not implemented in 0.9.3, right?

Thanks.

Nervehurter, 2019-03-26:
Hello @marcgpuig is the implementation of this feature still planned?
Thank you for your good work!

analog-cbarber, 2019-03-26:
You can already use the existing depth camera to mimic Lidar if you want. Just create a 3 or more depth cameras covering the full 360 view. That will give you a very accurate 360 degree pointcloud. If you want to mimic a Velodyne pattern, you would then sample points along the Velodyne scan patterns (probably better to do this on the depth maps rather than the pointcloud).

nsubiron, 2019-03-26:
We put this task on hold because of the technical difficulties simulating a rotating Lidar with a depth camera, but IMO we should still add the progress here at some point as it already gives you the 3D points from a depth image (like a Velarray sort of sensor maybe).

analog-cbarber, 2019-03-27:
Given that in 0.9 you can move your sensors dynamically, I imagine you could just have a single depth camera that you rotate from the client side, although you would probably have to use synchronous mode to get smooth results.

Nervehurter, 2019-03-28:
Thanks for all the input, I will take a Look at the depth Image solution. The Rotation ist a secondary Feature for me so it should work.

kimiya66, 2019-06-21:
how can I find the detected points with Lidar(detected vehicles)? Is this raw_data gives me location of detected vehicles? 
It is a buffer now pointing to memory I think. I do not have any idea how can I get information from that? i,e. how can I get the location of detected points or distance to them? 
how can I open this raw_data buffer?
some one please helping me. I would be thankful.

barbierParis, 2019-06-25:
Hey @marcgpuig ,
Did you manage to implement semantic segmentation for lidars ? And is your implementation the lidar_gpu branch?

FilippoCeffa, 2019-07-24:
The Z-buffer approach is a rasterization technique, and as such the quality of the result is dependent on the target buffer resolution. To approach the quality of the ray-casting approach, the resolution should be high enough so that the error introduced by discretization is negligible. How much error is acceptable in this kind of feature?
There is a second potential issue: after rasterization is done, the output will be a buffer, not a point cloud. A “crawl” step would be necessary to extract the relevant pixels to a 3D list of points. This might be expensive, especially if the resolution has been set quite high to address the aforementioned problem.
I was wondering if it would be possible to use GPU raytracing for this task, which would lead to the same precision as CPU ray-casting, with a much higher performance. Epic introduced RayCasting based on DXR in UE4.22, so it might be worth investigating how easy it would be to hook into the API.

analog-cbarber, 2019-07-24:
The z-buffer approach is equivalent to using the current depth sensor. We have been using the depth sensor quite a bit and it is more than accurate enough for this purpose. Real LIDAR sensors have much lower resolution and accuracy then you will get from this approach. Projecting depths to point clouds is not prohibitively expensive.  

The main reason for using ray tracing would be to develop more sophisticated LIDAR models that could account for reflections (e.g. off of water on the road).

FilippoCeffa, 2019-07-24:
I understand, thanks for the explanation.

cmpute, 2019-08-09:
Any progress that can be added? Really need a real shaped point cloud for algorithm testing, rotation simulation can be actually ignored since you can consider the sensor as solid state,as said in https://github.com/carla-simulator/carla/issues/779#issuecomment-476797748

Put this into high priority please! :nerd_face:

ZeKubiki, 2019-11-19:
Really looking forward to this sensor implementation. I work for a robotics company and anything approaching a high-fidelity lidar model would be invaluable. Even if it doesn't exactly match the scan patterns of any existing units.

germanros1987, 2020-03-19:
This is being addressed as part of the External Sensor Interface (ESI) subproject, so I am closing this issue. Please, open a new issue if needed.

closed 2020-03-19 by germanros1987

ZeKubiki, 2020-03-19:
What's the External Sensor Interface and is there an open issue for it?

cmpute, 2020-03-20:
Same question as above