carla-simulator/carla #780: How to get the control signal from autopilot in the new version of 0.9.0

xie9187, 2018-09-14:
Hi, thanks for developing such a good simulator for self-driving car.

I am trying to record the control signal in the autopilot mode by using the manul_control.py file. However I cannot find a way to read from autopilot, but only a function 'self._vehicle.set_autopilot(autopilot)' that sets the car in autopilot mode.

I did it successfully in the old version by reading from 'measurements.player_measurements.autopilot_control'. But since the API changed, I cannot find such a function anymore.

Is it still possible to read the control signal from the autopilot mode?

marcgpuig, 2018-09-18:
Hi @xie9187. 
The autopilot is about to change in 0.9.X. Please wait until a new release comes out.
We are working on it.

xie9187, 2018-09-18:
Thanks for telling me. Looking forward to that new release

closed 2018-09-18 by xie9187

peterkim95, 2020-09-14:
Any updates on this? I'm trying to get the control signal from autopilot in 0.9.X. Would the recommended way for now be to just use an older version?