carla-simulator/carla #782: Issues with Python manual_control.py (Carla 0.9.0 / Win)

grafandreas, 2018-09-16:
Hi all,

I have been successfully building Carla on Win but I am experiencing problems with the Python clients.

PythonAPI\manual_control.py:

* Connects successfully
* I can drive the car with the keys in the Python window
However:
* The callback for the camera sensor is never called, so nothing is displayed
I added this line to verify:
 def _parse_image(self, image):
        logging.debug('Image')
The camera object (a "sensor") seems to be created, though

PythonClient\manual_control.py:

Running this and pressing a key crashes the Carla Simulator (not only the Python client, but the Unreal Server).

Thank you very much for any feedback.

M-Hammod, 2018-09-17:
Hi, 
I had the same problem with the 0.9 release which got fixed 7 days ago (please check: https://github.com/carla-simulator/carla/pull/763) if you pull the latest changes and rebuild everything I believe this issue will be solved

grafandreas, 2018-09-17:
Hm, I updated and built yesterday. Could you tell me which commit you are on. I might have to install / clone from scratch, though.

marcgpuig, 2018-09-18:
@grafandreas @M-Hammod 
This problem was solved in #763.
Please try to rebuild your project. Also, make sure you rebuild the Python API.

morambro, 2018-09-21:
Hi, I compiled the latest version of Carla, but I have a similar issue on Linux: if I run manual_control.py locally on the same machine as Carla, all works fine and I get the video streamed. If however I run the client from a remote machine in LAN, Carla doesn't start streaming sensor data (e.g., using ports 200/2001, I see no traffic received/generated in both machines with tcpdump on 2001, while traffic on 2000 is regularly flowing).

closed 2018-12-01 by stale[bot]