carla-simulator/carla #785: Util/BuildTools/Linux.mk:54: recipe for target 'setup' failed make: *** [setup] Error 1

PygmalionChen, 2018-09-18:
I build the Unreal Engine successful. Download the carla, run the script.
But after running the following commond: 
make launch   # Compiles CARLA and launches Unreal Engine's Editor.
make package  # Compiles CARLA and creates a packaged version for distribution.

The error occur, how can I address the error?

Linux to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
-- Performing Test LIBCXX_SUPPORTS_FPIC_FLAG - Success
-- Adding Benchmark: vector_operations.bench.cpp
-- Adding Benchmark: algorithms.bench.cpp
-- Adding Benchmark: string.bench.cpp
-- Adding Benchmark: unordered_set_operations.bench.cpp
-- Adding Benchmark: filesystem.bench.cpp
-- Adding Benchmark: util_smartptr.bench.cpp
-- Adding Benchmark: stringstream.bench.cpp
-- Linker detection: GNU ld
-- Configuring incomplete, errors occurred!
See also "/home/pygmalionchen/PycharmProjects/carla/Build/llvm-5.0-build/CMakeFiles/CMakeOutput.log".
See also "/home/pygmalionchen/PycharmProjects/carla/Build/llvm-5.0-build/CMakeFiles/CMakeError.log".
Util/BuildTools/Linux.mk:54: recipe for target 'setup' failed
make: *** [setup] Error 1

PygmalionChen, 2018-09-18:
The CMakeError.log:

Source file was:
int main() { return 0; }
Performing C++ SOURCE FILE Test LIBCXX_SUPPORTS_EHSC_FLAG failed with the following output:
Change Dir: /home/pygmalionchen/PycharmProjects/carla/Build/llvm-5.0-build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/ninja" "cmTC_38e3c"
[proxychains] DLL init: proxychains-ng 4.12-git-1-gbb3df1e
[1/2] Building CXX object CMakeFiles/cmTC_38e3c.dir/src.cxx.obj
FAILED: /usr/bin/clang++-5.0     -fPIC -fvisibility-inlines-hidden -Werror=date-time -std=c++11 -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers  -Wno-long-long -Wcovered-switch-default -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -ffunction-sections -fdata-sections -DLIBCXX_SUPPORTS_EHSC_FLAG  -nodefaultlibs   -EHsc -o CMakeFiles/cmTC_38e3c.dir/src.cxx.obj -c src.cxx
[proxychains] DLL init: proxychains-ng 4.12-git-1-gbb3df1e
[proxychains] DLL init: proxychains-ng 4.12-git-1-gbb3df1e
clang: error: unknown argument: '-EHsc'
ninja: build stopped: subcommand failed.

nsubiron, 2018-09-18:
Hi @PygmalionChen, which platform are you using? it seems CMake is not able to find the configuration for your system.

nsubiron, 2018-09-18:
Our dependencies require CMake 3.9 or later. It is a weird error nevertheless, may be something wrong with your installation.

PygmalionChen, 2018-09-19:
I install the CMake 3.12 and make the project success, thank you!

closed 2018-09-19 by marcgpuig

geekhut, 2019-01-10:
> I install the CMake 3.12 and make the project success, thank you!

Hi how did you install CMake 3.12? I am having problems with this too.

WellingtonLobatoJunior, 2020-01-05:
> > I install the CMake 3.12 and make the project success, thank you!
> 
> Hi how did you install CMake 3.12? I am having problems with this too.

https://vitux.com/how-to-install-cmake-on-ubuntu-18-04/

uoyssim, 2020-03-31:
> > I install the CMake 3.12 and make the project success, thank you!
> 
> Hi how did you install CMake 3.12? I am having problems with this too.

Installing cmake using apt install can only install cmake version 3.10. 

remove the currently installed cmake.
$ sudo apt remove cmake

You need to download the latest cmake from cmake.org. Then install it.
https://cmake.org/download/

I install the CMake 3.17 and make the project success :)