carla-simulator/carla #792: Issue with opening Carla

andrster, 2018-09-20:
[image]
getting this issue when trying to use carla. Any help would be great.

marcgpuig, 2018-09-20:
Hi @andrster.
Can you provide more information?
How many versions of Unreal do you have installed? Which version are you using? Are you rebuilding CARLA?

andrster, 2018-09-20:
I believe 4.18 and yes I was rebuilding carla.

nsubiron, 2018-09-21:
If you press "Yes" does it give any error? I've seen this happening sometimes, usually pressing yes it recompiles and launches the editor just fine.

andrster, 2018-09-21:
Yes it won't recompile and to try manually, which also won't work.  Also my symbol for the project looks off and make rebuild in carla gives that error
[image]

andrster, 2018-09-21:
and this happens on Rebuild.sh
[image]

andrster, 2018-09-21:
issue resolved

closed 2018-09-21 by andrster