carla-simulator/carla #793: Issue about do experiments from start point to end point

chensg1919, 2018-09-20:
https://carla.readthedocs.io/en/latest/benchmark_creating/

From the link I know how to do experiments from start point to end point. However, the vehicle can only go straight. One test is [140,134]. But the vehicle cannot arrive at 134 because it will collide and stop. In PythonAPI, I see there is only foward agent. Is there any way to drive a car from start point to end point even there is a turn?

chensg1919, 2018-09-20:
And also, I want to know whether I could replace coordinate number with (x,y) coordinate, such as replace 140 with (662,1305)

chensg1919, 2018-09-26:
Any solution to solve it?

felipecode, 2018-09-26:
Hey @chensg1919 We are currently have an agent to do that
The agent is for version 0.8.4. It is placed on the client_agent_improving_pedestrians branch.
The agent is called command follower, you can see an example of it in the control_example.py python code.

chensg1919, 2018-09-26:
@felipecode 
Sry, I'm now using 0.8.4, but I cannot find this branch. Could you tell me the specific location?

felipecode, 2018-09-27:
https://github.com/carla-simulator/carla/tree/client_agents_improving_pedestrians

and the code ( example)

https://github.com/carla-simulator/carla/blob/client_agents_improving_pedestrians/PythonClient/control_example.py

The actual agent code

https://github.com/carla-simulator/carla/blob/client_agents_improving_pedestrians/PythonClient/carla/agent/command_follower.py

chensg1919, 2018-09-27:
@felipecode Thank you. I have downloaded this branch, but there are still some questions. 
Firstly, I cannot run Package.sh. "fatal: Not a git repository (or any of the parent directories): .git"
And also, when I run driving_benchmark_example.py, it shows me ImportError: No module named 'pid_controller'.

How to fix it?

closed 2018-12-03 by stale[bot]