carla-simulator/carla #795: Simulation is very slow

mhusseinsh, 2018-09-21:
I am running CARLA 0.8.4 on Windows 10
I am running the manual_control python client, however the simulation is very very slow, it is like performing maybe 2 frames per second or something like that

Is there any way to speed up the simulation, or maybe is there a way to use the GPU, if this can help ?

Note: I already reduced the size of the server
`.\CarlaUE4.exe -carla-server -windowed -ResX=1 -ResY=1`
but it seems it is not also helping, the pygame itself is very slow

Any tips/ideas ?

chankim, 2018-10-11:
Hi, I have the same issue. (maybe slower). I tried it ubuntu 16.04, with two GPUs. I compiled, ran "CarlaUE4.sh -windowed" and ran "manual_control.py". As I press WSAD keys, I can see the car moving in the big window. But the small window (from inside the car) is updated so slowly and catches up some seconds later. What is wrong?

zxbnjust, 2018-10-29:
u should change the drive of your video card.use the independent one.

closed 2019-01-04 by stale[bot]