carla-simulator/carla #797: Is it possible to use custom datasets in CARLA?

anubhav0fnu, 2018-09-21:
Hello,

Is there a way we can give custom video dataset as input to CARLA and see how a self-driving car actually drives in the custom environment?
The dataset could be real-world, such as [BDD100K](https://github.com/ucbdrive/bdd-data) or [Cityscapes](https://www.cityscapes-dataset.com/).

I searching for an open-source simulator wherein I can plug-play a already trained self-driving car model and test it on different datasets to see how well the autonomous model performs.?

I didn't find a way to it using documentation.

marcgpuig, 2018-09-26:
Hi @anubhav0fnu 
I'm not sure if I understand you. CARLA itself does not provide any car AI that you can use in any of these datasets. If you are planning to create and train an (external) agent with CARLA, then you will be able to use this agent to see how it drives using the dataset you want. 
But this last part is up to you, it has nothing to do with CARLA :)

Cheers.

anubhav0fnu, 2018-09-26:
@marcgpuig , so, if I can create a self-driving agent by myself, is it possible to incorporate a real-world dataset in the simulator, for example, the one I mentioned above?

marcgpuig, 2018-09-28:
@anubhav0fnu What I can't understand is what do you mean with
> incorporate a real-world dataset in the simulator

Do you mean another map like @Romanenko-Serhii suggested?

closed 2018-12-04 by stale[bot]