carla-simulator/carla #800: Change in Server

FarzanehJafari, 2018-09-22:
Hi,
I am going to change the server quiet a bit. I would like both of client and server control vehicles with keyboard. What do I  do about this problem? Could you help me?
Thanks,

FarzanehJafari, 2018-09-24:
Hi,
I added a parameter to Player_Carla_Measurements and sent to client. Client Could read this data, but after a second, Carla crashed. 
What do you think about this problem?

FarzanehJafari, 2018-09-24:
[
[image]
](url)

nsubiron, 2018-09-25:
Hi @FarzanehJafari,

We provide a Python script that allows controlling the player vehicle with the keyboard: `manual_control.py`, wouldn't that work for you?

FarzanehJafari, 2018-09-25:
Hi, this python code woked fo me, but I want driving both with client and sometimes with server sach as change weathers with server. Does it work possible? and Which classes should be change on server side?Another problem is about benchmark driving. python code doesnt work for me and there isnt run_driving_benchmark in directory. Thanks a lot,

nsubiron, 2018-09-25:
Having both inputs simultaneously is a bit tricky since they will interfere with each other, as both inputs are being applied to the vehicle continuously every frame.

Take a look at WheeledVehicleController.cpp, there is where we suppress the input from the simulator window when the client is active.

FarzanehJafari, 2018-09-25:
which function of wheeledvehiclecontroller.cpp is doing this work? I dont found it.

nsubiron, 2018-09-25:
The `Set*Input()` functions at the bottom of the file,

```cpp
if (bAllowUserInput && !IsAutopilotEnabled()) {
  // set input here.
}
```
if Carla is running in server mode, `bAllowUserInput` is false.

FarzanehJafari, 2018-09-25:
I set constant value for bAllowUserInput  = true, but yet, I couldnt
control vehicle with server. 🙁🙁
Whats your suggestion about it? Are there any parameters, I should be
changed?

nsubiron, 2018-09-25:
As I mention, probably the input coming from the client is overwriting the input from the server, you'd need to find a way of merging both inputs.

FarzanehJafari, 2018-09-25:
Excuse me for my questions. You tired to answer them and I am sorry about
it. last question, which classes can I see this Input functions for server
and standalone modes?
Thanks a lot for your help.😳

nsubiron, 2018-09-25:
Hi @FarzanehJafari, no worries. The input from the keyboard comes from that class, WheeledVehicleController.cpp; and the input from the client-side comes from `DataRouter::ApplyVehicleControl` in DataRouter.h. But every frame you get control from both sides (including 0.0 if the key is not pressed), so the last one writing is the one that actually applies. The tricky part is how to combine these two so the vehicle control still makes sense.

FarzanehJafari, 2018-09-30:
Hi, Thank you very much. I could use from Client and Server for control Vehicle. but, I have another question about add a parameter to measurements and send to client. I add an integer parameter to void FCarlaEncoder::Encode() function and run. but I have this Error

FarzanehJafari, 2018-09-30:
[Desktop/45946081-9929df00-bfa4-11e8-8290-f1c0f0bd427d.png](url)

FarzanehJafari, 2018-09-30:
[image]

nsubiron, 2018-09-30:
That's not an error, those are variables optimized by the compiler. If you change the configuration to debug in Visual Studio you'll see those variables.

FarzanehJafari, 2018-09-30:
I dont understand exactly? could you explain about it for me with more
details? Thanks

nsubiron, 2018-09-30:
Sorry, those are general programming issues we cannot give support to.

I would recommend reading more about how the compiler optimizes code, and how Visual Studio debugging works, e.g.

- https://docs.microsoft.com/en-us/visualstudio/debugger/how-to-debug-optimized-code?view=vs-2017
- https://docs.microsoft.com/en-us/visualstudio/debugger/how-to-set-debug-and-release-configurations?view=vs-2017

Hope those links are a good starting point.

FarzanehJafari, 2018-09-30:
Thaks a lot

   
   - https://docs.microsoft.com/en-us/visualstudio/debugger/how-to-debug-optimized-code?view=vs-2017
   - https://docs.microsoft.com/en-us/visualstudio/debugger/how-to-set-debug-and-release-configurations?view=vs-2017

closed 2018-12-06 by stale[bot]