carla-simulator/carla #801: Obtaining traffic light state for each frame

SoumiDas, 2018-09-23:
Hello,

I was wondering how do I **obtain the Traffic light information** which gets **displayed on the screen in the simulator** while it's running. The **print_same_line** function in **client_example.py** which prints these measurement doesn't have the traffic light state information printed in code as I saw.

So can you please help me out with this?

Thanks!

nsubiron, 2018-09-25:
Hi @SoumiDas,

In the list of [non-player agents](https://carla.readthedocs.io/en/stable/measurements/#non-player-agents-info) we include the traffic lights and their states (on API 0.8.x). Unfortunately, it is not stated anywhere which traffic light is currently applying to you, (basically because we are still looking for an standardized way to do so), but you could try to deduce it by their distance from the player.

SoumiDas, 2018-09-25:
Thanks a lot @nsubiron !
So, how is the information of traffic light on the bottom left of the screen displayed while the simulator is running? Is it calculated in the same way as you mentioned here and the state is shown accordingly?

nsubiron, 2018-09-25:
No, inside the simulator we have extra info that is not available in the client. There are trigger boxes that notify the vehicle when there is a traffic light.

closed 2018-12-01 by stale[bot]

Pranav21091996, 2019-11-28:
Hi, @nsubiron is it possible to get the distance of the car from the next traffic light? I did not find any other function other than getting a traffic light state.

SoumiDas, 2019-11-28:
Hi,

You can get the transform/position of the next traffic light just as we got
the state.

You have the agent vehicle's transform too.

Taking these two points, you can get the distance of the traffic light!