carla-simulator/carla #805: 0.9.0 linux, Carla crash when call manual_control.py

wobistdu, 2018-09-26:
Play in editor or standalone both have this issue.
Steps
1. Play the editor with default level Town01
2. Call python manual_control.py 
3. UE editor crashes when the python script goes to carla/client.py:158
```
155  	        # Send new episode request.
156  	        pb_message = carla_protocol.RequestNewEpisode()
157  	        pb_message.ini_file = str(carla_settings)
158 B->	        self._world_client.write(pb_message.SerializeToString())
159  	        # Read scene description.
160  	        data = self._world_client.read()
161  	        if not data:
162  	            raise RuntimeError('failed to read data from server')
163  	        pb_message = carla_protocol.SceneDescription()
```
I checked, now carla uses rpc server, and define the actions in FTheNewCarlaServer::FPimpl::BindActions. 
But the client python script is sending in old format? How should I work around this? Do I need to compile with CARLA_COMPILE_CARLASERVER_LEGACY?
Thanks a lot
The crash call stack is 
`#0  0x00007fffec4ac428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007fffec4ae02a in __GI_abort () at abort.c:89
#2  0x00007ffff7505b4d in abort_message () from /home/tiankuan/UnrealEngine_4.19/Engine/Binaries/Linux/libUE4Editor-Core.so
#3  0x00007ffff7505c41 in default_terminate_handler() () from /home/tiankuan/UnrealEngine_4.19/Engine/Binaries/Linux/libUE4Editor-Core.so
#4  0x00007ffff75027f3 in std::__terminate(void (*)()) () from /home/tiankuan/UnrealEngine_4.19/Engine/Binaries/Linux/libUE4Editor-Core.so
#5  0x00007ffff7504436 in __cxa_throw () from /home/tiankuan/UnrealEngine_4.19/Engine/Binaries/Linux/libUE4Editor-Core.so
#6  0x00007ffddfa6239f in clmdep_msgpack::v1::adaptor::convert<std::__1::tuple<signed char, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, clmdep_msgpack::v2::object>, void>::operator()(clmdep_msgpack::v2::object const&, std::__1::tuple<signed char, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, clmdep_msgpack::v2::object>&) const () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#7  0x00007ffddfa62324 in clmdep_msgpack::v1::operator>><std::__1::tuple<signed char, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, clmdep_msgpack::v2::object> >
    () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#8  0x00007ffddfa622ed in clmdep_msgpack::v1::object::convert<std::__1::tuple<signed char, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, clmdep_msgpack::v2::object> > () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#9  0x00007ffddfa546e8 in std::__1::tuple<signed char, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, clmdep_msgpack::v2::object>& clmdep_msgpack::v2::object::convert<std::__1::tuple<signed char, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, clmdep_msgpack::v2::object> >(std::__1::tuple<signed char, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, clmdep_msgpack::v2::object>&) const () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#10 0x00007ffddfa526e9 in rpc::detail::dispatcher::dispatch_notification(clmdep_msgpack::v2::object const&, bool) ()
   from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#11 0x00007ffddfa524f7 in rpc::detail::dispatcher::dispatch(clmdep_msgpack::v2::object const&, bool) () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#12 0x00007ffddfa7c052 in rpc::detail::server_session::do_read()::$_1::operator()(std::__1::error_code, unsigned long) const::{lambda()#1}::operator()() const ()
   from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#13 0x00007ffddfa7bfa5 in void clmdep_asio::clmdep_asio_handler_invoke<rpc::detail::server_session::do_read()::$_1::operator()(std::__1::error_code, unsigned long) const::{lambda()#1}>(rpc::detail::server_session::do_read()::$_1::operator()(std::__1::error_code, unsigned long) const::{lambda()#1}&, ...) () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#14 0x00007ffddfa7bf07 in void clmdep_asio_handler_invoke_helpers::invoke<rpc::detail::server_session::do_read()::$_1::operator()(std::__1::error_code, unsigned long) const::{lambda()#1}, {lambda()#1}>(rpc::detail::server_session::do_read()::$_1::operator()(std::__1::error_code, unsigned long) const::{lambda()#1}&, {lambda()#1}&) ()
   from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#15 0x00007ffddfa7bde6 in clmdep_asio::detail::completion_handler<rpc::detail::server_session::do_read()::$_1::operator()(std::__1::error_code, unsigned long) const::{lambda()#1}>::do_complete(clmdep_asio::detail::task_io_service*, clmdep_asio::detail::task_io_service_operation*, std::__1::error_code const&, unsigned long) ()
   from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#16 0x00007ffddfa6b4e7 in clmdep_asio::detail::task_io_service_operation::complete(clmdep_asio::detail::task_io_service&, std::__1::error_code const&, unsigned long) ()
   from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#17 0x00007ffddfa74900 in clmdep_asio::detail::task_io_service::do_run_one(clmdep_asio::detail::scoped_lock<clmdep_asio::detail::posix_mutex>&, clmdep_asio::detail::task_io_service_thread_info&, std::__1::error_code const&) () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#18 0x00007ffddfa7450f in clmdep_asio::detail::task_io_service::run(std::__1::error_code&) () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#19 0x00007ffddfa65c28 in clmdep_asio::io_service::run() () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#20 0x00007ffddfa654b8 in rpc::server::async_run(unsigned long)::$_0::operator()() const () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#21 0x00007ffddfa653ed in void std::__1::__invoke_void_return_wrapper<void>::__call<rpc::server::async_run(unsigned long)::$_0&>(rpc::server::async_run(unsigned long)::$_0&) ()
   from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#22 0x00007ffddfa65239 in std::__1::__function::__func<rpc::server::async_run(unsigned long)::$_0, std::__1::allocator<rpc::server::async_run(unsigned long)::$_0>, void ()>::operator()() ()
   from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#23 0x00007ffddfa773db in std::__1::function<void ()>::operator()() const () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#24 0x00007ffddfa76e84 in void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, std::__1::function<void ()> > >(void*) () from /home/tiankuan/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
#25 0x00007ffff7bc16ba in start_thread (arg=0x7ffd86bfd700) at pthread_create.c:333
#26 0x00007fffec57e41d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
`

wobistdu, 2018-09-26:
The client is sending message using the format form carla_server_pb2.py.

DESCRIPTOR = _descriptor.FileDescriptor(
  name='carla_server.proto',
  package='carla_server',
  syntax='proto3',
  serialized_pb=_b('\n\x12\x63\x61rla_server.proto\x12\x0c\x63\x61rla_server\"+\n\x08Vector3D\x12\t\n\x01x\x18\x01 \x01(\x02\x12\t\n\x01y\x18\x02 \x01(\x02\x12\t\n\x01z\x18\x03 \x01(\x02\"6\n\nRotation3D\x12\r\n\x05pitch\x18\x01 \x01(\x02\x12\x0b\n\x03yaw\x18\x02 \x01(\x02\x12\x0c\n\x04roll\x18\x03 \x01(\x02\"\x92\x01\n\tTransform\x12(\n\x08location\x18\x01 \x01(\x0b\x32\x16.carla_server.Vector3D\x12/\n\x0borientation\x18\x02 \x01(\x0b\x32\x16.carla_server.Vector3DB\x02\x18\x01\x12*\n\x08rotation\x18\x03 \x01(\x0b\x32\x18.carla_server.Rotation3D\"a\n\x0b\x42oundingBox\x12*\n\ttransform\x18\x01 \x01(\x0b\x32\x17.carla_server.Transform\x12&\n\x06\x65xtent\x18\x02 \x01(\x0b\x32\x16.carla_server.Vector3D\"\x80\x01\n\x06Sensor\x12\n\n\x02id\x18\x01 \x01(\x07\x12\'\n\x04type\x18\x02 \x01(\x0e\x32\x19.carla_server.Sensor.Type\x12\x0c\n\x04name\x18\x03 \x01(\t\"3\n\x04Type\x12\x0b\n\x07UNKNOWN\x10\x00\x12\n\n\x06\x43\x41MERA\x10\x01\x12\x12\n\x0eLIDAR_RAY_CAST\x10\x02\"}\n\x07Vehicle\x12*\n\ttransform\x18\x01 \x01(\x0b\x32\x17.carla_server.Transform\x12/\n\x0c\x62ounding_box\x18\x04 \x01(\x0b\x32\x19.carla_server.BoundingBox\x12\x15\n\rforward_speed\x18\x03 \x01(\x02\"\x80\x01\n\nPedestrian\x12*\n\ttransform\x18\x01 \x01(\x0b\x32\x17.carla_server.Transform\x12/\n\x0c\x62ounding_box\x18\x04 \x01(\x0b\x32\x19.carla_server.BoundingBox\x12\x15\n\rforward_speed\x18\x03 \x01(\x02\"\x94\x01\n\x0cTrafficLight\x12*\n\ttransform\x18\x01 \x01(\x0b\x32\x17.carla_server.Transform\x12/\n\x05state\x18\x02 \x01(\x0e\x32 .carla_server.TrafficLight.State\"\'\n\x05State\x12\t\n\x05GREEN\x10\x00\x12\n\n\x06YELLOW\x10\x01\x12\x07\n\x03RED\x10\x02\"Q\n\x0eSpeedLimitSign\x12*\n\ttransform\x18\x01 \x01(\x0b\x32\x17.carla_server.Transform\x12\x13\n\x0bspeed_limit\x18\x02 \x01(\x02\"\xe5\x01\n\x05\x41gent\x12\n\n\x02id\x18\x01 \x01(\x07\x12(\n\x07vehicle\x18\x02 \x01(\x0b\x32\x15.carla_server.VehicleH\x00\x12.\n\npedestrian\x18\x03 \x01(\x0b\x32\x18.carla_server.PedestrianH\x00\x12\x33\n\rtraffic_light\x18\x04 \x01(\x0b\x32\x1a.carla_server.TrafficLightH\x00\x12\x38\n\x10speed_limit_sign\x18\x05 \x01(\x0b\x32\x1c.carla_server.SpeedLimitSignH\x00\x42\x07\n\x05\x61gent\"%\n\x11RequestNewEpisode\x12\x10\n\x08ini_file\x18\x01 \x01(\t\"\x80\x01\n\x10SceneDescription\x12\x10\n\x08map_name\x18\x03 \x01(\t\x12\x33\n\x12player_start_spots\x18\x01 \x03(\x0b\x32\x17.carla_server.Transform\x12%\n\x07sensors\x18\x02 \x03(\x0b\x32\x14.carla_server.Sensor\"/\n\x0c\x45pisodeStart\x12\x1f\n\x17player_start_spot_index\x18\x01 \x01(\r\"\x1d\n\x0c\x45pisodeReady\x12\r\n\x05ready\x18\x01 \x01(\x08\"^\n\x07\x43ontrol\x12\r\n\x05steer\x18\x01 \x01(\x02\x12\x10\n\x08throttle\x18\x02 \x01(\x02\x12\r\n\x05\x62rake\x18\x03 \x01(\x02\x12\x12\n\nhand_brake\x18\x04 \x01(\x08\x12\x0f\n\x07reverse\x18\x05 \x01(\x08\"\xd1\x04\n\x0cMeasurements\x12\x14\n\x0c\x66rame_number\x18\x05 \x01(\x04\x12\x1a\n\x12platform_timestamp\x18\x01 \x01(\r\x12\x16\n\x0egame_timestamp\x18\x02 \x01(\r\x12J\n\x13player_measurements\x18\x03 \x01(\x0b\x32-.carla_server.Measurements.PlayerMeasurements\x12.\n\x11non_player_agents\x18\x04 \x03(\x0b\x32\x13.carla_server.Agent\x1a\xfa\x02\n\x12PlayerMeasurements\x12*\n\ttransform\x18\x01 \x01(\x0b\x32\x17.carla_server.Transform\x12/\n\x0c\x62ounding_box\x18\x0c \x01(\x0b\x32\x19.carla_server.BoundingBox\x12,\n\x0c\x61\x63\x63\x65leration\x18\x03 \x01(\x0b\x32\x16.carla_server.Vector3D\x12\x15\n\rforward_speed\x18\x04 \x01(\x02\x12\x1a\n\x12\x63ollision_vehicles\x18\x05 \x01(\x02\x12\x1d\n\x15\x63ollision_pedestrians\x18\x06 \x01(\x02\x12\x17\n\x0f\x63ollision_other\x18\x07 \x01(\x02\x12\x1e\n\x16intersection_otherlane\x18\x08 \x01(\x02\x12\x1c\n\x14intersection_offroad\x18\t \x01(\x02\x12\x30\n\x11\x61utopilot_control\x18\n \x01(\x0b\x32\x15.carla_server.ControlB\x03\xf8\x01\x01\x62\x06proto3')
)

nsubiron, 2018-09-26:
Hi @wobistdu,

You are using the 0.8.x API with the 0.9.0 simulator, they're no longer compatible. The 0.9.0 is a work-in-progress showing the new API we are implementing.

- The new client (0.9.x) is located in _PythonAPI_ folder.
- The old client (0.8.x) is located in _PythonClient_ folder.

Sorry for the confusion, we are in middle of the transition between the two APIs and documentation is still a bit messy.

harlowja, 2018-09-27:
One seems to be msgpack based (0.9) and the other (0.8) protobuf based right?

wobistdu, 2018-09-27:
@nsubiron  Thanks for the quick reply, I am able to PIE now.
One more question, by default the project is using ATheNewCarlaGameModeBase, but I compared it with previous CarlaGameMode, the new one does not load weather settings, does it mean it doesn't support dynamic weather now?

wobistdu, 2018-09-28:
For now I switch back to CarlaGameMode to make it work.

nsubiron, 2018-09-28:
> @harlowja One seems to be msgpack based (0.9) and the other (0.8) protobuf based right?

Yes, exactly. Carla 0.9 is using @rpclib for all the communications except the sensor data, sensor data is sent raw to increase speed.

@wobistdu The weather and quality level settings are temporarily disabled in the new game mode, we want to give it some thought to the API that will control them. We will add them soon.

AliBaheri, 2018-10-03:
I have created a new folder and download an stable 8 version, and repeated the process, but still the car floats between the earth and sky :(

closed 2018-11-05 by nsubiron