carla-simulator/carla #808: How to load my own map by the carla simulator?

chengjinjack, 2018-09-27:
Hello,everyone!
According to official document( `https://carla.readthedocs.io/en/latest/map_customization/`),I modify  an existing map (defaultmap.umap)which is in the unreal/CarlaUE4/content/maps/ by the Unreal Engine Editor,and python client program can connect  to it .I want to load this map by the carla simulator without running in  the Unreal Engine 4,so I do something,for example,I run this:
`./CarlaUE4.sh  /Game/Maps/Defaultmap`,but in fact this is not successful .How can I do this?

Romanenko-Serhii, 2018-09-27:
Hello, @chengjinjack 
I think you need to include your own map to the project for exporting.
You can do it in Unreal Engine it by:
Settings -> Project Settings -> in search field put "map" and find "Project - Packaging" -> add in "List of maps to include in a packaged build" path to your map. 
Then you must again youse Package project in tab File

chengjinjack, 2018-09-29:
Thank you very much,@Romanenko-Serhii
I successfully implemented it,but when I run the carla_ros_bridge package ,for example ,`roslaunch carla_ros_bridge client_with_rviz.launch
`,I run into a issue that there are not defaultmap.png file in the PythonClient/carla/planner/. I know that showing road map in the rivz need some files,such as in the term of town01.umap,Town01.png Town01.txt Town01Central.png Town01Lanes.png are needed,is there a way to generate therse files ?
I hope everyone can help me.

Sakrust, 2018-10-01:
These files are generated by the CarlaMapGenerator, you can trigger it inside the editor, make sure to deselect 'generate roads' and to tick the checkbox save road map to disk.

chengjinjack, 2018-10-02:
Hello,@Sakrust.
Thank you for your patience.
Through the method you said，I generate a file which is similar to Town01Lanes.png ,but three other documents haven't been generated,what should I do next?

chengjinjack, 2018-10-08:
Hello,everyone!
Is there a solution to the above problem?

chengjinjack, 2018-10-25:
Hi,@Sakrust
is there a way to solve the question?

Sakrust, 2018-10-25:
The .umap is created by unreal itself. If you save your map, this umap file should be  somewhere in Content/Maps. And what other files are missing?

chengjinjack, 2018-10-26:
Hi,@Sakrust
when I run the carla_ros_bridge package ,for example ,roslaunch carla_ros_bridge client_with_rviz.launch ,I run into a issue that there are not defaultmap.png file in the PythonClient/carla/planner/. I know that showing road map in the rivz need some files,such as in the term of town01.umap,Town01.png Town01.txt Town01Central.png Town01Lanes.png are needed,is there a way to generate therse files ?
This problem has been bothering me for a long time.

closed 2019-01-01 by stale[bot]

NebulousNeo, 2019-09-05:
This don't work for me, how can i save a customized town in Unreal Engin as a .pcd file or in such way that i can start it with the command like ./CarlaUE4.sh /home/map....

NebulousNeo, 2019-09-05:
okay i found a solution, it was the fault of the .egg file in the folder there wasn't the right .egg file version (carla-0.9.5-py2.7-linux-x86_64.egg instead of should be carla-0.9.6-py2.7-linux-x86_64.egg). This problem occur because i had two carla folder the first one was the old carla version(0.9.5) and the new one(0.9.6 build from source). So i copied from folder /home/xxx/Test/carla/PythonAPI/carla/dist/ (folder of 0.9.6) the egg files to /home/xxx/yyy/zzz/Carla/CARLA_0.9.5/PythonAPI/carla/dist (this is 0.9.5 where i also execute my pythonAPI). after that it worked and i could conntect to the carla world.