carla-simulator/carla #809: Relase GIL at python API boundaries

harlowja, 2018-09-27:
From the looks of it the boost python (new) layer is not releasing the GIL for python.

This makes having a responsive UI (using pyglet or pygame) pretty hard to accomplish (especially say when a timeout to the carla server happens).

An easy way to test this is to setup a server, then try to run a client, then after the client connects, kill the server and the app will usually freeze up.

Something like https://stackoverflow.com/a/41168814 is probably needed...

nsubiron, 2018-09-28:
Yes, that freeze-up you mention is definitively a bug.

But I'm not sure whether this is a problem of releasing the Python GIL or just a matter of time-outs and job cancellations in the networking client code. I'm not sure if it is a good idea to let Python continue if we are not able connect, we can block while we try, but making sure we don't wait forever. I will investigate this.

chankim, 2018-10-11:
Hi, What should I do when the client window(pygame window) is not killable. I can't kill the client, including the client window. The server is killed by Alt-F4 but not the client. What process should I kill? (I mean from ps eux list..)

nsubiron, 2018-10-11:
There is a deadlock currently in master, should be fixed in #856. I'm afraid the only way is killing it from the OS.

chankim, 2018-10-11:
Hi, I upgraded the pygame package and the server-client is working well. (for manual_control.py, with --autopilot). For now, I don't have a situation that I have to kill them, so I can use alt-F4. I'm just testing CARLA, and don't know how to use it yet. I hope the tutorial-like document can be provided soon for 0.9.0. That seems to be a nice tool! Thanks!