carla-simulator/carla #810: Including LibCarla in a C++ project

harlowja, 2018-09-27:
Due to how LibCarla (especially the client) builds its own version of boost and rpclib I am wondering if this code was made to be included by other C++ applications. It does not really appear that it is possible to use it as is (due to the bundled libraries and such) so I am wondering if there needs to be some kind of documentation saying yes this is possible, or no it isnt.

nsubiron, 2019-07-01:
We have now an example [Examples/CppClient](https://github.com/carla-simulator/carla/tree/master/Examples/CppClient).

closed 2019-07-01 by nsubiron

jemishGhoghari, 2024-03-29:
> We have now an example [Examples/CppClient](https://github.com/carla-simulator/carla/tree/master/Examples/CppClient).

Hello @nsubiron,

How to build this Example on Windows? or Do you have any lead how to do that?

Any Help appreciated!

Thanks