carla-simulator/carla #812: Error when "make launch"

xwixwi1893, 2018-09-27:
I followed the tutorial to build CARLA on Ubuntu 16.04. Everything is fine until I compiles the CARLA. The error is like:
"[6/32] Building CXX object LibCarla/cm...reaming/detail/tcp/ServerSession.cpp.o
FAILED: /usr/bin/clang++-5.0  -DBOOST_ERROR_CODE_HEADER_ONLY -I../../LibCarla/cmake/../source -I../boost-1.67.0-install/include -I../rpclib-2.2.1-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wno-missing-braces -stdlib=libc++ -I/home/xwi/carla/Build/llvm-5.0-install/include/c++/v1 -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wno-missing-braces -stdlib=libc++ -I/home/xwi/carla/Build/llvm-5.0-install/include/c++/v1   -O3 -DNDEBUG -MD -MT LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/streaming/detail/tcp/ServerSession.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/streaming/detail/tcp/ServerSession.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/streaming/detail/tcp/ServerSession.cpp.o -c ../../LibCarla/source/carla/streaming/detail/tcp/ServerSession.cpp
In file included from ../../LibCarla/source/carla/streaming/detail/tcp/ServerSession.cpp:7:
In file included from ../../LibCarla/cmake/../source/carla/streaming/detail/tcp/ServerSession.h:10:
../../LibCarla/cmake/../source/carla/Time.h:9:10: fatal error: 'boost/date_time/posix_time/posix_time_types.hpp' file not found
#include <boost/date_time/posix_time/posix_time_types.hpp>"

What can I do to deal with it?

nsubiron, 2018-09-28:
That's weird. You are using version 0.9.0 right? Can you check if the following file was properly created?

    Build/boost-1.67.0-install/include/boost/date_time/posix_time/posix_time_types.hpp

xwixwi1893, 2018-10-01:
Thank you. I checked the folder and found that nothing is built under boost-1.67.0-install. However, I re-download Carla and build it again. It works this time.

closed 2018-11-05 by nsubiron