carla-simulator/carla #813: Error when `make LibCarla`

harlowja, 2018-09-28:
Unable to build this on ubuntu 16.04.

Some build information:

```
Linux josh-XPS-15-9560 4.15.0-34-generic #37~16.04.1-Ubuntu SMP Tue Aug 28 10:44:06 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

cmake version 3.12.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).

rm -rfv Build/ && make LibCarla 
....
....
BuildLibCarla.sh: Building LibCarla "Client" configuration.
-- The C compiler identification is Clang 5.0.0
-- The CXX compiler identification is Clang 5.0.0
-- Check for working C compiler: /usr/bin/clang-5.0
-- Check for working C compiler: /usr/bin/clang-5.0 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-5.0
-- Check for working CXX compiler: /usr/bin/clang++-5.0 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building libcarla version 0.9.0-111-g09df66a-dirty
-- Configuring done
-- Generating done
-- Build files have been written to: /home/josh/Dev/carla/Build/libcarla-client-build
[1/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/Actor.cpp.o
[2/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/ActorAttribute.cpp.o
[3/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/ActorBlueprint.cpp.o
[4/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/BlueprintLibrary.cpp.o
[5/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/Client.cpp.o
[6/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/Image.cpp.o
[7/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/World.cpp.o
[8/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/rpc/Server.cpp.o
[9/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/streaming/detail/Dispatcher.cpp.o
[10/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/streaming/detail/Token.cpp.o
[11/30] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/streaming/detail/tcp/Client.cpp.o
FAILED: /usr/bin/clang++-5.0  -DBOOST_ASIO_ENABLE_BUFFER_DEBUGGING -DBOOST_ERROR_CODE_HEADER_ONLY -I../../LibCarla/cmake/../source -I../boost-1.67.0-install/include -I../rpclib-2.2.1-libstdcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wno-missing-braces -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wno-missing-braces   -g -MD -MT LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/BlueprintLibrary.cpp.o -MF LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/BlueprintLibrary.cpp.o.d -o LibCarla/cmake/client/CMakeFiles/carla_client_debug.dir/__/__/source/carla/client/BlueprintLibrary.cpp.o -c ../../LibCarla/source/carla/client/BlueprintLibrary.cpp
In file included from ../../LibCarla/source/carla/client/BlueprintLibrary.cpp:7:
In file included from ../../LibCarla/cmake/../source/carla/client/BlueprintLibrary.h:10:
In file included from ../../LibCarla/cmake/../source/carla/Iterator.h:9:
In file included from ../boost-1.67.0-install/include/boost/iterator/transform_iterator.hpp:11:
In file included from ../boost-1.67.0-install/include/boost/iterator/iterator_adaptor.hpp:12:
In file included from ../boost-1.67.0-install/include/boost/iterator/iterator_categories.hpp:23:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/iterator:64:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/ostream:38:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/ios:42:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/ios_base.h:41:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/locale_classes.h:40:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/string:41:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/allocator.h:46:
In file included from /usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/x86_64-linux-gnu/c++/5.4.0/bits/c++allocator.h:33:
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/ext/new_allocator.h:120:23: error: no matching constructor for initialization of 'std::pair<const std::__cxx11::basic_string<char>, carla::client::ActorBlueprint>'
        { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
                             ^   ~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/alloc_traits.h:530:8: note: in instantiation of function template specialization '__gnu_cxx::new_allocator<std::pair<const std::__cxx11::basic_string<char>, carla::client::ActorBlueprint> >::construct<std::pair<const std::__cxx11::basic_string<char>, carla::client::ActorBlueprint>, const std::__cxx11::basic_string<char> &, const carla::rpc::ActorDefinition &>' requested here
        { __a.construct(__p, std::forward<_Args>(__args)...); }
              ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/hashtable_policy.h:1955:28: note: in instantiation of function template specialization 'std::allocator_traits<std::allocator<std::pair<const std::__cxx11::basic_string<char>, carla::client::ActorBlueprint> > >::construct<std::pair<const std::__cxx11::basic_string<char>, carla::client::ActorBlueprint>, const std::__cxx11::basic_string<char> &, const carla::rpc::ActorDefinition &>' requested here
            __value_alloc_traits::construct(__a, __n->_M_valptr(),
                                  ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/hashtable.h:1526:30: note: in instantiation of function template specialization 'std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<const std::__cxx11::basic_string<char>, carla::client::ActorBlueprint>, true> > >::_M_allocate_node<const std::__cxx11::basic_string<char> &, const carla::rpc::ActorDefinition &>' requested here
        __node_type* __node = this->_M_allocate_node(std::forward<_Args>(__args)...);
[... 22 lines cut ...]
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_pair.h:124:12: note: candidate constructor template not viable: requires single argument '__p', but 2 arguments were provided
        constexpr pair(const pair<_U1, _U2>& __p)
                  ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_pair.h:150:12: note: candidate constructor template not viable: requires single argument '__p', but 2 arguments were provided
        constexpr pair(pair<_U1, _U2>&& __p)
                  ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_pair.h:155:9: note: candidate constructor template not viable: requires 3 arguments, but 2 were provided
        pair(piecewise_construct_t, tuple<_Args1...>, tuple<_Args2...>);
        ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_pair.h:206:9: note: candidate constructor template not viable: requires 4 arguments, but 2 were provided
        pair(tuple<_Args1...>&, tuple<_Args2...>&,
        ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_pair.h:127:17: note: candidate constructor not viable: requires 1 argument, but 2 were provided
      constexpr pair(const pair&) = default;
                ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_pair.h:128:17: note: candidate constructor not viable: requires 1 argument, but 2 were provided
      constexpr pair(pair&&) = default;
                ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/stl_pair.h:108:26: note: candidate constructor not viable: requires 0 arguments, but 2 were provided
      _GLIBCXX_CONSTEXPR pair()
                         ^
1 error generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:51: recipe for target 'LibCarla' failed
```

harlowja, 2018-09-28:
[output.txt.zip](https://github.com/carla-simulator/carla/files/2426850/output.txt.zip)

Full log

harlowja, 2018-09-28:
[output.txt](https://github.com/carla-simulator/carla/files/2426851/output.txt)

Full log attached.

harlowja, 2018-09-28:
It's almost like it's not looking for the libcxx that is being built?

nsubiron, 2018-09-28:
I think the problem is that it's compiling against gcc 5 stdlib headers, but it should compile against gcc 7 instead. That's why in the documentation we have g++-7 as dependency, with that installed CMake should pick the right headers and runtime libs.

Note that unlike the server, the client side doesn't link against libc++, but libstdc++.

harlowja, 2018-09-28:
Ah, kk, let me try that and see how it goes.

AliBaheri, 2018-10-01:
Is there any update for this issue?
I have the same problem here, and the solution provided here dose not work in my case.

Manojbhat09, 2018-10-16:
yes @AliBaheri fixed it 💯 

https://github.com/carla-simulator/carla/issues/875

harlowja, 2018-10-17:
Ya, I tried it with the right g++7 and all seemed better.

Thanks all for the help (and the fix)! :)