carla-simulator/carla #814: CarlaClient is not thread-safe

Ajk4, 2018-09-28:
Hi, 

I'm reporting this issue for Carla 0.8.4 - it's possible that it's fixed in 0.9.x. In that case please ignore this issue.

In my use-case I'm using python `threading` library. Each thread connects to it's own Carla server and has it's own CarlaClient object.

I noticed, that `player_measurements` are interleaved. Meaning that Thread 1 might sometimes receive `player_measurements` from Thread 2.

nsubiron, 2018-09-28:
A single CarlaClient object is not thread-safe, but I can't see anything in this class that wouldn't allow you to have two instances of this class living independently in different threads, as long as they are communicating with simulators in different addresses. Perhaps the protobuf message is not thread-safe? Do you have a minimal example to reproduce this?

marcgpuig, 2018-09-28:
Hi @Ajk4, just to ensure, are you using different ports for different Carla servers?

Ajk4, 2018-09-28:
I am using different ports. 

Thanks for replies. If I have time I'll try to prepare minimal-example.

pawel-ziecina, 2018-10-17:
I'm observing also interleaved frames on carla 0.9.0 while using carla server and client on different machines. It might be related with this issue. Steps to reproduce:

1. Start carla server 0.9.0 on nodeA: `./CarlaUE4.sh -windowed -carla-server -carla-no-hud`
2. Create tunnel to forward streaming port (2001) from nodeA to nodeB: `ssh -L2001:nodeA:2001 nodeA` (I guess that it is required because server returns that streaming server listens on 127.0.0.1:2001 and there is no possibility to change this)
3. On nodeB run attached manual `manual_control.py` (it adds saving frames to png copied from example.py): `python manual_control.py --host nodeA -a`

Observe: after some time frames staring to be displayed in not proper sequence; in `_images` directory there will be frames with consecutive frames number

[manual_control.zip](https://github.com/carla-simulator/carla/files/2486543/manual_control.zip)

nsubiron, 2018-10-17:
Hi @pawel-ziecina, can that be a different issue? --> #815

pawel-ziecina, 2018-10-17:
on v0.8.4 it is not observable

closed 2018-12-23 by stale[bot]