carla-simulator/carla #815: Client receives interleaved sensor messages

nsubiron, 2018-09-28:
With 0.9.x versions, the streaming server may interleave messages while sending, e.g. image of frame 3 arrives before images of frames 1 and 2. This happens if the server-client communication is slower than the sensor.

This happens because the server currently enqueues a message again when a writing operation is currently being executed by another thread. This opens a window for a newly created message to enter the queue before this one.

Shall we ignore messages while the socket is busy or shall we keep a message queue?

  * If we ignore messages, as it's usually done in video-games, we will reduce lag at the cost of missing some of the messages. This will remove too the problem of having a message queue able to grow indefinitely.
  * If we keep a queue, we may introduce a small performance overhead (possibly negligible, should be tested) but we reduce the number of messages lost and we can put a limit on the number of messages that are stored at any given time.

analog-cbarber, 2018-09-28:
You should probably do either depending on an option setting. There will be some users who don't want any dropped frames, while others may just want the best performance.

nsubiron, 2018-10-03:
We opted for discarding the messages if the connection/client is too slow. Once we add the synchronous mode to the new API this shouldn't be a problem since you can force the simulation to wait for the client each frame.

jhanhiro, 2018-10-08:
Hi,
I am facing this problem of interleaved sensor messages. I tested the issue#815 branch, but using cameras with the basic client examples (manual_driving.py and example.py) does not work. For example, executing the example.py with add_a_camera=True produces the following error:

$ python example.py 
client version: 0.9.0-158-g0f16fed
server version: 0.9.0-158-g0f16fed
spawning vehicle 'vehicle.mini.cooperst' with 4 wheels
Vehicle(id=34, type=vehicle.mini.cooperst)
Traceback (most recent call last):
  File "example.py", line 115, in <module>
    main(add_a_camera=True, enable_autopilot=True)
  File "example.py", line 88, in main
    camera_bp = blueprint_library.find('sensor.camera')
IndexError: _Map_base::at

Without adding a camera the example runs fine.

nsubiron, 2018-10-08:
Hi @jhanhiro,

That's actually a different issue, we changed the name of the cameras but forgot to update the examples, just replace

    camera_bp = blueprint_library.find('sensor.camera')

with

    camera_bp = blueprint_library.find('sensor.camera.rgb')

it will be fixed when these commits go to master. Thanks for reporting!

commit 7e6a19dc (2018-10-03): Close #815, discard message if the connection is too slow
commit fa19f029 (2018-10-19): Close #815, discard message if the connection is too slow
PR #854 (2018-10-23): Fix client receives interleaved sensor messages
#### Description

Requires #750.

See description and comments in #815.

I have removed the part that "re-posts" a message when the socket is busy. Now it simply ignores the message. It is a bit drastic but reduces latency on client considerably when connection is slow, plus this won't be a problem once we enable the synchronous mode.

I have also tweaked the unit tests to reflect that we can miss messages.

Fixes #815.

#### Where has this been tested?

  * **Platform(s):** Ubuntu 16.04 and Windows 10
  * **Unreal Engine version(s):** 4.19

#### [... cut]
closed 2018-10-23 by nsubiron