carla-simulator/carla #817: "make launch" dose not work!

AliBaheri, 2018-09-28:
I am trying to install CARLA on Ubonto.

The `./Update.sh` goes smooth...But I am not able to run `make launch`. Specifically, I am facing with the following error after running `make launch`:

`rm: cannot remove 'llvm-5.0-source/projects/libcxx/test/std/input.output/iostreams.base/std.ios.manip/floatfield.manip/hexfloat.pass.cpp': Permission denied
`

Could you please help me to resolve this issue?

Thanks

nsubiron, 2018-10-01:
Make sure you have write permissions in the folder you are building Carla.

AliBaheri, 2018-10-01:
Thanks for your response.

In fact, I have followed all instructions in the documentation and after having write permissions I am facing with the following message when I run `make launch`:

`1 error generated.

  "clang++" -c -x c++ -fPIC -m64 -pthread -O3 -Wall -Wno-inline -fPIC -std=c++14 -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_ALL_NO_LIB=1 -DBOOST_NUMPY_SOURCE -DNDEBUG -I"." -I"/home/abaheri/miniconda3/include/python3.5" -I"/home/abaheri/miniconda3/lib/python3.5/site-packages/numpy/core/include" -o "bin.v2/libs/python/build/clang-linux-5.0/release/threading-multi/numpy/ufunc.o" "libs/python/src/numpy/ufunc.cpp"

...failed clang-linux.compile.c++.without-pth bin.v2/libs/python/build/clang-linux-5.0/release/threading-multi/numpy/ufunc.o...
...skipped <pbin.v2/libs/python/build/clang-linux-5.0/release/threading-multi>libboost_numpy35.so.1.67.0 for lack of <pbin.v2/libs/python/build/clang-linux-5.0/release/threading-multi>numpy/dtype.o...
...skipped <pstage/lib>libboost_numpy35.so.1.67.0 for lack of <pbin.v2/libs/python/build/clang-linux-5.0/release/threading-multi>libboost_numpy35.so.1.67.0...
...skipped <pstage/lib>libboost_numpy35.so for lack of <pstage/lib>libboost_numpy35.so.1.67.0...
...failed updating 66 targets...
...skipped 12 targets...
...updated 18 targets...
Util/BuildTools/Linux.mk:54: recipe for target 'setup' failed
make: *** [setup] Error 1

All previous commands for build in Linux in the instruction has gone pretty smooth.

Thanks for your help.

`

AliBaheri, 2018-10-01:
Even updating the  Cmake to the latest version did not help.
Any help would be greatly appreciated.

Manojbhat09, 2018-10-15:
Were you able to fix it @AliBaheri ? I am facing the same issue/

AliBaheri, 2018-10-15:
@Manojbhat09 
I was able to resolve the issue by installing the latest version of Cmake.

Try:

`pip install cmake=3.12`

Hope it helps.

Manojbhat09, 2018-10-16:
Yes it works, thank you!. Anyways its now stuck at building libcarla 

This is the error i am getting:
Setup.sh: llvm-5.0 already installed.
Setup.sh: boost-1.67.0 already installed.
Setup.sh: rpclib-2.2.1 already installed.
Setup.sh: googletest-1.8.0 already installed.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server" configuration.
[22/46] Building CXX object LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o
FAILED: /usr/bin/clang++-5.0  -DBOOST_ERROR_CODE_HEADER_ONLY -I../../LibCarla/cmake/../source -I../boost-1.67.0-install/include -I../rpclib-2.2.1-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wno-missing-braces -stdlib=libc++ -I/mnt/d/manojfiles/carla/Build/llvm-5.0-install/include/c++/v1 -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wno-missing-braces -stdlib=libc++ -I/mnt/d/manojfiles/carla/Build/llvm-5.0-install/include/c++/v1   -O3 -DNDEBUG -MD -MT LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o -c ../../LibCarla/source/carla/Buffer.cpp
In file included from ../../LibCarla/source/carla/Buffer.cpp:1:
../../LibCarla/cmake/../source/carla/Buffer.h:204:42: error: no member named 'is_const_buffer_sequence' in namespace 'boost::asio'
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                            ~~~~~~~~~~~~~^
../../LibCarla/cmake/../source/carla/Buffer.h:204:67: error: 'T' does not refer to a value
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                  ^
../../LibCarla/cmake/../source/carla/Buffer.h:203:24: note: declared here
    template <typename T>
                       ^
../../LibCarla/cmake/../source/carla/Buffer.h:204:71: error: member 'value' declared as a template
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                      ^
../../LibCarla/cmake/../source/carla/Buffer.h:204:76: error: expected ';' at end of declaration list
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                           ^
                                                                           ;
../../LibCarla/cmake/../source/carla/Buffer.h:97:9: error: use of undeclared identifier 'ReuseThisBuffer'
        ReuseThisBuffer();
        ^
../../LibCarla/source/carla/Buffer.cpp:7:16: error: out-of-line definition of 'ReuseThisBuffer' does not match any declaration in 'carla::Buffer'
  void Buffer::ReuseThisBuffer() {
               ^~~~~~~~~~~~~~~
6 errors generated.

please let me know if this issue can be resolved

sh-shu, 2018-11-06:
> Yes it works, thank you!. Anyways its now stuck at building libcarla

 This is exactly what I met. Have you resolved this problem?

closed 2019-01-12 by stale[bot]