carla-simulator/carla #818: Image difference between 'Python Client‘ and 'SceneCaptureToDiskCamera'

yangzuyuanhao, 2018-09-28:
Hi,

Just found images generated by python client (attachment 1) and images generated by Mustang SceneCaptureToDiskCamera (attachment 2) are very different. Note that these two pictures are under same environment settings/ same weather mode/ same location. 
So, these two pictures generated from different data source? 
[image]
[image]

nsubiron, 2018-09-30:
Hi @yangzuyuanhao,

I think the difference is that the cameras added by the client get some adjustment depending on the weather(*) while the SceneCaptureToDiskCamera is only there for debugging purposes and does not get such adjustments.

Which version are you using? Nevertheless I would like to take a look to make sure.

- - -

(*) This is done to match the view of the simulator, you would expect scene captures producing images with same quality and effects as in the game main camera but this is not the case. We have reported to the Unreal team but they told us the only workaround is to "manually" adjust these cameras to match the main camera view.

yangzuyuanhao, 2018-10-01:
@nsubiron,

Thanks for you help!
I am using Carla 0.8 with UE4 4.18.
So, if I want to stick with python client, but I want to make some adjustment on the camera mode, how can I do it? I am not sure how does the image raw data generated..

SpakMiguel, 2018-10-03:
I understand then that pictures captured by save-images-to-disk are taken by a camera that is located exactly at the point manually indicated on the python scripts? Is there any weather condition that does not move the position of the camera on the Python Client?

nsubiron, 2018-10-03:
In 0.8.x versions, the camera position remains unchanged (relative to the vehicle) for the whole run; the weather cannot change positions. To find out the sensor position you need to calculate it based in the vehicle's current position and the transform of the sensor relative to the vehicle.

yangzuyuanhao, 2018-10-04:
@nsubiron,
If I understand you correctly, the image generated by python client still based on class SaveImageToDiskCamera? So, If I want to change the image effect (like contrast, saturation, etc), I should 
change the SaveImageToDiskCamera c++ code, right?

nsubiron, 2018-10-05:
The class is actually its parent, `ASceneCaptureCamera`. That's the one that uses the client, you can change the code there to modify the object, but bear in mind most of the configuration is done at runtime when the camera is loaded, based on post-process effect and current weather.

closed 2018-10-09 by yangzuyuanhao

1453042287, 2018-11-13:
@yangzuyuanhao did you solve the problem?