carla-simulator/carla #823: How can I create the a geometry plane  with 4 (or more)known points in points coordinate?

quanheng123, 2018-10-01:
Hey,  I have the question as title described,

For example, I 've already know the 4 (x, y, z=0)or more points in coordinate, I want to create the non-standard plane with these points, and after that I can make it become static meshes and add it in carlamapgenerator and MapRoad, make it recognised?
How can I create this non-standard Plane?

SpakMiguel, 2018-10-01:
In order to do that I believe you have to model your plane with a 3D modelling tool such as Blender or even Unreal. Then, you should open the CARLA project on the Unreal editor, import the plane and rebuild the whole project. 

Correct me if I am wrong, I would like to do something similar.

marcgpuig, 2018-10-01:
Hey @quanheng123.
Not sure if this it's what you are searching for, but in my mesh generation tests, I've found [SiggiG's ProceduralMeshes](https://github.com/SiggiG/ProceduralMeshes) to be very useful.

SpakMiguel, 2018-10-01:
What @marcgpuig has posted can be helpful to you in order to create geometry in Unreal because if you do not use tools like that you can only create geometry with relative characteristics and that is not what you are looking for. The other way is as you say: importing already created geometry as fbx format from an external tool and rebuild CARLA.

quanheng123, 2018-10-01:
Thanks, that's very helpful.

closed 2018-10-01 by quanheng123