carla-simulator/carla #824: build Carla 0.8.2 on Win 10 system  with no error but get almost blank start screen

matthew-rococo, 2018-10-01:
i have build the carla-0.8.2 version from the source on my win10 computer according to  the instruction of the document step by step and finished with no error (even no warning).  The build command i issued is "make launch" , when build progress finished,  it launched a very strange screen almost with nothing on it (except the base ground ) , as the following pic:
[image]
  i also tried  "make launch-editor" , just the same result.
  i then use  UE4 Editor opening  the CarlaUE4.uproject  at the directory "Unreal\CarlaUE4\",  the editor gives out a lot of errors said 'can't find file for asset ...", but these files is actually exist at the said directory. The errors screen is like the following pic:
[image]

 i used build this version with UE4  4.18.3 by the old way (maybe 3 months ago), it is some kind of complicated, but everything is OK.     

  can anyone help me?

marcgpuig, 2018-10-01:
Hi @matthew-rococo 
Did you download the assets and extracted them into `G:\carla-strm\Unreal\CarlaUE4\Content\Carla\`?
(Notice the `Carla` folder at the end)

matthew-rococo, 2018-10-02:
YES, i did download the content and carefully check the version and the location. maybe my carla version is.wrong? but i have checked thhis with the.changelog file,it seems ok. is there anyway else to assure this?

adarsh9975, 2018-10-02:
@matthew-rococo Content (assets) placement of carla 0.8.2 and 0.9.0 are different. The automated  build process was made for 0.9.0 version which seems incompatible with other version hence you get that message. Tried building 0.8.2 this morning. Didn't workout for me, so switched to 0.9.0

marcgpuig, 2018-10-02:
@matthew-rococo it really look like you have a wrong assets version or the path is not correct...
@adarsh9975 

## Getting the content
Here:

    \Util\ContentVersions.txt

you will find the download link for each version of Carla.

The automated installation gives you the correct link in 0.8.X after the setup. Also, the 0.9.0 automated installation gives the correct link just after the setup, but it gets lost under all the log. A workarround in 0.9.0 is running `make setup`, so the installation only does the setup process (if needed) and you will see the download link you need.

## Content placement
#### 0.8.X
    \Unreal\CarlaUE4\Content\
#### 0.9.X
    \Unreal\CarlaUE4\Content\Carla\

_Notice we added a `Carla` folder at the end!_

Hope this helps

matthew-rococo, 2018-10-02:
thanks a lot. you are absolutly right: i uesed the wrong carla version number. i thought i am using the stable version , actually i am using the master version:version 0.92.
i made this mistake because i use git destop and it is very easy to switch between different version and i am not familer with it.
btw, can you tell me why the content of 0.9x is only about 3 gb while the content of 0.8x is about  10g? it seems 0.9x has more UE4 assets.

marcgpuig, 2018-10-02:
Hey @matthew-rococo.

Typical of git :)

Don't worry, there is no lack of content. The thing is that all the content was revised and reorganized. With a lot of work, our artists have removed all the assets that we are not using, like obsolete objects, really big or unused textures and so on.

closed 2018-10-02 by marcgpuig